Arnold-n/P1P2MQTT #27: [home-assistant] Energy statistics not available

Timple, 2022-08-29:
I wanted to add my heatpump to the energy dashboard but it was not in the dropdown list.

As requested per this link: https://www.home-assistant.io/more-info/statistics/ I know opened this issue pointing this project to this link: https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

Is the `state_class` something that can be communicated via mqtt?

Arnold-n, 2022-08-29:
Thanks tim, hopefully this works in just released P1P2-bridge-esp8266 v0.9.18. Coding is done ad-hoc, as it will be coded differently in P1P2MQTT.

Timple, 2022-08-30:
Great, thanks!

Would it make sense to attached these releases as binaries to the releases tab in github?

Arnold-n, 2022-08-30:
Just did! Thanks for the suggestion.

Timple, 2022-08-31:
Awesome!
I'll verify asap (which is two weeks unfortunately)

closed 2022-08-31 by Timple

Timple, 2022-09-12:
Works! Thanks 👍