Arnold-n/P1P2MQTT #29: [Home Assistant] some topics fail discovery

fishbone-git, 2022-09-04:
Home Assistant logs attached below. I suspect te issue is an MQTT buffer problem as it seems to be the very long topic names that get cut off at the end, so the discovery topic fails to parse due to missing closing } at least
[halogs.txt](https://github.com/Arnold-n/P1P2Serial/files/9485553/halogs.txt)

Arnold-n, 2022-09-04:
Hi @fishbone-git, indeed it is the MQTT buffer size, please upgrade to 0.9.18 which increases this (HA_VALUE_LEN) from 300 to 600 to resolve this issue.

fishbone-git, 2022-09-04:
in progress

fishbone-git, 2022-09-04:
The buffer problem appears solved, but there is a new problem.  I'm using HA 2022.9.0b4.  All topics appear in MQTT broker, and in th discovery topic.  HA is having a problem with a number of them, and so they are no longer available entities in HA after upgrading to 0.9.18. Hard  to check all topics, but the ones generating the error all seem to have state_class="".  I do not believe this is allowed by HA.  When I manually publish a discovery topic to remove state_class="", it immediately creates an entity in HA
[halogs.txt](https://github.com/Arnold-n/P1P2Serial/files/9485604/halogs.txt)

Arnold-n, 2022-09-04:
You're fast and fully correct! Sorry, I should have advised to jump to the newest version: 0fd7519 solves this too.

fishbone-git, 2022-09-04:
I just spotted that too :).  I guess let's close this and either I'll see if I can now get a working build or wait until next binaries are published. Thx!

Arnold-n, 2022-09-04:
I'll make them available soon - was waiting for the climate and water heater integration to be finished and tested.

closed 2022-09-04 by Arnold-n

Arnold-n, 2022-09-04:
Not as official release, but v0.9.20 images (for E_series) are [here](https://github.com/Arnold-n/P1P2Serial/tree/main/Firmware_images).

Arnold-n, 2022-09-05:
The images have been replaced as the ATmega OTA upgrade didn't work properly due to a BSP issue on my side. The current Eseries images should be fine.

fishbone-git, 2022-09-06:
Thanks.  I'll try the images tonight and see if there is a difference.  I did finally manage to get a build on arduinoIDE that seems stable