Arnold-n/P1P2MQTT #30: Interested in buying p1p2-esp interface

stomko11, 2022-11-03:
As title says. Do you have any available? 
Can I have both this and daikin controller connected at the same time? How do I wire them?

Arnold-n, 2022-11-03:
Hi @stomko11 please send me email for details.

Not sure if you refer to the main (room) controller or the Daikin LAN/WiFi adapter.

All controllers are wired in parallel on P1 and P2.
You must have one "main controller" which is the room controller/thermostat.
You may have one (in some systems two) auxiliary controllers active, such as a Daikin LAN adapter, or P1P2Serial, but not both.

P1P2Serial can also work in listen-only mode, in which case it can always be connected to P1/P2, but only if there not too many consumers on the bus (we do not know what the max bus load is).

stomko11, 2022-11-04:
Hello. I actually failed to find your email address anywhere - where do I send you email?
I am using only one controller currently, it is not serving as thermostat, for that I use external on/off contact. So my daikin controller serves only to configure my Altherma air to water heatpump. I would like to keep that available in technical room, and wire that in parallel somehow. Main goal is to be able to turn on/off DHW/Heating. Would I just connect it on the pins inside controller?
[image]

Arnold-n, 2022-11-04:
Indeed you can wire it in parallel, near the controller in your image, or near the heat pump, as long as there is WiFi (or if there is ethernet with the extra ethernet adapter). As the wires in your picture are quite thick I would split them outside the unit, and make the last wire thinner (= more flexible). My e-mail is a bit hidden in the README and [in the code](https://github.com/Arnold-n/P1P2Serial/blob/main/P1P2Serial.cpp#L3).