Arnold-n/P1P2MQTT #38: ºC vs °C

Timple, 2022-12-01:
You might notice by eye a slight difference in the degree symbols.

And as they are different, the cannot be compared or plotted in the same graph by home-assistant.

I wanted to put in a pull-request with a fix, but I cannot seem to find the code snippet that causes this. The 'wrong' item is `S1_DHW_Comfort_Status`

Arnold-n, 2022-12-07:
Hi @Timple,
Sorry for the delay. The symbol changed on 7 September from ºC (U+00BA) to °C (U+00B0) (symbol is defined in P1P2_Daikin_ParameterConversion_EHYHB.h line 164 (code will be replaced in P1P2MQTT), but I do not understand why you would see two different symbols as there is only one location defining it, unless a retained HA MQTT discovery message from before 7 September remains. Also, I do not recognize S1_DHW_Comfort_Status, I do not think it originates from P1P2Serial, do you know where this comes from?

Timple, 2022-12-08:
The message originated from `P1P2/P/155/S/1/DHW_Comfort_Status`. But you are right, it seems a retained HA MQTT message. I can simply remove it.

Thank you for looking into it!

closed 2022-12-08 by Timple