Arnold-n/P1P2MQTT #56: Altherma 3 RW (EHBH08EF9W) interpretation

trantoriana, 2023-01-04:
Hi Arnold,

Been watching the data on MQTT and home assistant for two days now, and most of the times the mqtt data and HA sensors make sense.

One sensor I am missing though is the operation. In espaltherma this is the 'IU Operation Mode' which is delivered as Stop or Heating.
I am unable to find the right counterpart in p1p2serial.
Presumably one of the OnOff ones, but none are chaning their state. For instance 'S0_Heating_OnOff' is always 1 and does not change.
Any pointers for me on the meaning on some of the sensors?

Also, I am heavily using home assistant in my setup and see it is being worked on.
One thing that may make intergration better is to prefix all HA sensors with 'P1P2_' because the current names are too generic and not unique. Prefixing may help with the finding of the right sensors.

Thanks, Trant

trantoriana, 2023-01-05:
> For instance 'S0_Heating_OnOff' is always 1 and does not change.

ok, 'S0_Heating_OnOff' is the general OnOff for the heatpump.

Looking for another ones... for instance the one when the madoka is displaying heating using the blue light.
Is there a counterpart or should the pump and rising leaving temperature be used for that?

T

Arnold-n, 2023-01-05:
Hi @trantoriana , thanks for your help!

Actually S0_Heating_OnOff is not directly related to "IU operating mode" as it represents the request from the main controller to switch heating on or off (0 indicates source is main controller).

I think that
-"IU Operation Mode" corresponds to S2_DHW_Heating_Gas_Status_related (0=Off, 1=Heating, 4=DHW, 5=Heating+DHW), however this value is only available if an auxiliary controller is present ("L1" or another device), and it is a combination of
-S1_Heating_OnOff or S1_Valve_Zone_Main (both seem to change together, 0 for no heating, 1 for heating, not sure if there are situations where these differ and which would then correspond to IU Operation Mode), and
-S1_Valve_DHW_Tank (0 for no DHW, 1 for DHW).

Can you check how these values are reported on your system?

trantoriana, 2023-01-05:
> Can you check how these values are reported on your system?

Bit inconsistant:

* S/1/Heating_OnOFF_1  is in MQTT but not in HA 
* S/1/Valve_Zone_Main   is in MQTT but not in HA
* S/1/Valve_DHW_Tank   is in MQTT but not in HA 
* S/2/DHW_Heating_Gas_Status_related not available in HA

I will keep an eye on the values in MQTT so I can let you know

Thanks.

trantoriana, 2023-01-05:
Both heating onoff 1 and Valve main zone switch outside of the heating schedule between 0 and 1, Does not seem to indicate heating :-)

Arnold-n, 2023-01-06:
Thanks, and I can add these sensors to HA in the next version.

W.r.t. your earlier request to prefix sensors with P1P2_, I am hesitant as the names are often too long to be shown entirely, and they are already part of a single device. They should be unique, also because of the S1_ -style prefixes.

trantoriana, 2023-01-07:
>  They should be unique, also because of the S1_ -style prefixes.

True.

However, speaking from a user perspective when creating automations or template sensors, home assistant auto complete is a bit dumb and since I cannot remember the exact p1p2 sensor to pick, starting with p1p2 would reduce the pickable list to only those devices I want to pick from.

Starting with S1 or A9 or A8 or P1 is not unique enough (for me) to pick the sensors, besides... depending on personal naming convention I may already have sensors starting with S1 or P1, commonly modbus devices for instance.

How about adding a configurationable item where user can set the prefix themselves?

> as the names are often too long to be shown entirely

I have yet to come in conflict with the max sensor name length in home assistant, and my fronius sunspec (hacs) integration generates some awfully long sensor names, example:

* sensor.power_photovoltaics_fronius_power_flow_0_http_mountdoom_untrusted_fenced_*********_at 

(domain redacted :-) )

so if it is the HA sensor name you worry about, then we should be on the save side.

S/1/Valve_main_Zone seems to indeed be exactly the same as the blue 'heating' light on the modoka thermostat. So that is one I would need.

Slowly I will start looking into controlling the heatpump but I am very modest. I only need:

* Turn heatpump on or off (basically controlling S0_Heating_OnOff)
* Set the target heating water temperatur (S0_Target_Temperature_Heating_Zone_Main_Abs)
* Set the main thermostat room temperatur (s0_target_temperature_room)

All the rest is just to showoff to my friends with the same heatpump :-)

Thanks, Fred

Arnold-n, 2023-01-07:
> How about adding a configurationable item where user can set the prefix themselves?

Understand your ratio, and good idea, I will add it to a next version.
 
> S/1/Valve_main_Zone seems to indeed be exactly the same as the blue 'heating' light on the modoka thermostat. So that is one I would need.

So it indeed makes sense to add these valves to HA too, consider it done in next version.

Arnold-n, 2023-01-15:
A prefix (defined in P1P2_Config.h) of "P1P2_" is added to all entities if useSensorPrefixHA is true; this can be set using the 8th argument of the B command or (if flash is erased or EEPROM signature is changed) via INIT_USE_SENSOR_PREFIX_HA in P1P2_Config.h. For backwards compatibility the prefix is not set by default, but if there is interest I can make it default for new installations.

The valves should be visible in HA too now.

trantoriana, 2023-01-15:
I guess once I manage to get a new firmware in the esp32 I will be able to verify, still I assume this addresses and close the issue.

Thanks, Trant

closed 2023-01-15 by trantoriana