Arnold-n/P1P2MQTT #69: Smart grid option

Timple, 2023-05-31:
Our Daikin heatpump has two relays for smart grid connectivity.

I am not sure if this is even within the protocol, but could this project support the smart grid features as well?

Arnold-n, 2023-06-04:
I think it is possible within the P1/P2 protocol, and it is on my todo-list; I purchased an BRP069A61 to see to what extent this can be done via P1/P2, but this will take some time. I hope P1/P2 enables for more precise control than is possible via the SG relay inputs.

bausi2k, 2023-06-05:
Maybe the hints on this side will help you further: https://community.openhab.org/t/how-to-integrate-daikin-altherma-lt-heat-pump/16488/14

I use the BRP069A61 with a shelly that closes the pins 
[image]
and I send HTTP Request to the BRP069A61 with the form data to change the value the _allowed power consupmtion_. This is a bit hacky, but it works. 
The get the formdata, just open an debug window in your browser and see what is going on, when you change values. 
But be careful, this only works for an hour, after starting the BRP069A61. After that, the Website will show you a HTTP500 on connection.

Timple, 2023-06-05:
> I hope P1/P2 enables for more precise control than is possible via the SG relay inputs.

And the P1/P2 wires already run into my house, so saves me some new cabling as wireless connection near/inside the heatpump is hard.

I might look into your suggestions as well @bausi2k 

Main purpose for now is to start heating the boiler water when te sun is high. To prevent my solar pannels to drop out due to too high voltage on the main :smile: . I need some consumes and this seems like a useful one.

bausi2k, 2023-06-05:
For that case, the BRP069A61 works like a charm. Even if you have a static value for the allowed consumption eg. 3500W the heat-pump takes the power and heats up the boiler(if room buffering is disabled) to the max allowed temperature(60degrees Celsius in my example) event though the desired temperature is only set to 47°C.

olfon, 2023-06-07:
> and I send HTTP Request to the BRP069A61 with the form data to change the value the _allowed power consupmtion_. This is a bit hacky, but it works. The get the formdata, just open an debug window in [...]

Would you mind to share the needed HTTP request? My LAN adapter always runs into issues after power loss, it is a hazzle to get it running again. Many thanks in advance :)

bausi2k, 2023-06-09:
This is my complete HTML File - find the form in the lower third

```<!DOCTYPE html>
<html>
	<head>
		<title></title>
	</head>
	<body>
		<form method="post" style="display:initial">
			<h1>
				Smart Grid
			</h1>
			<table id="table_smartgrid" width="40%">
				<tr style="display:table-row">
					<td>
						<p align="right">
							Pulse meter setting
						</p>
					</td>
					<td>
						<select name="DA_SG_PULSE_METER_SETTING">
							<option value="0" selected>
								No meter
							</option>
							<option value="4">
								100 pulse/kWh
							</option>
							<option value="5">
								1000 pulse/kWh
							</option>
						</select>
					</td>
				</tr>
				<tr style="display:table-row">
					<td>
						<p align="right">
							Electrical heaters allowed
						</p>
					</td>
					<td>
						<input type="radio" name="DA_SG_ELECTRICAL_HEATERS_ALLOWED" value="0">No <input type="radio" name="DA_SG_ELECTRICAL_HEATERS_ALLOWED" value="1" checked>Yes
					</td>
				</tr>
				<tr style="display:table-row">
					<td>
						<p align="right">
							Room buffering allowed
						</p>
					</td>
					<td>
						<input type="radio" name="DA_SG_ROOM_BUFFERING_ALLOWED" value="0" checked>No <input type="radio" name="DA_SG_ROOM_BUFFERING_ALLOWED" value="1">Yes
					</td>
				</tr>
				<tr style="display:table-row">
					<td>
						<p align="right">
							Static power limitation
						</p>
					</td>
					<td>
						<select name="DA_SG_STATIC_POWER_LIMITATION">
							<option value="500">
								0.5kW
							</option>
							<option value="1000">
								1kW
							</option>
							<option value="1500">
								1.5kW
							</option>
							<option value="2000">
								2kW
							</option>
							<option value="2500" selected>
								2.5kW
							</option>
							<option value="3000">
								3kW
							</option>
							<option value="3500">
								3.5kW
							</option>
							<option value="4000">
								4kW
							</option>
							<option value="4500">
								4.5kW
							</option>
							<option value="5000">
								5kW
							</option>
							<option value="5500">
								5.5kW
							</option>
							<option value="6000">
								6kW
							</option>
							<option value="6500">
								6.5kW
							</option>
							<option value="7000">
								7kW
							</option>
							<option value="7500">
								7.5kW
							</option>
							<option value="8000">
								8kW
							</option>
							<option value="8500">
								8.5kW
							</option>
							<option value="9000">
								9kW
							</option>
							<option value="9500">
								9.5kW
							</option>
							<option value="10000">
								10kW
							</option>
						</select>
					</td>
				</tr>
				<tr>
					<td>
						<!-- leer -->
					</td>
					<td>
						<input type="submit" name="DA_SG_SUBMIT" value="Submit">
					</td>
				</tr>
			</table>
		</form>
	</body>
</html>
```
[Daikin BRP069A61.txt](https://github.com/Arnold-n/P1P2Serial/files/11709431/Daikin.BRP069A61.txt)

nicx, 2023-09-21:
@Arnold-n any idea when this smart grid feature will be implemented?

@bausi2k am I right that the heat-pump will always be included and its not possible to use only the backup heater? I would like to conserve the heat pump and use only the backup heater for buffering heat in the water tank.

Arnold-n, 2023-09-21:
@nicx this may take some time as other things are higher priority (HA integration and COP calculation).

But why would you prefer the backup heater over the more efficient heat pump?

nicx, 2023-09-21:
> @nicx this may take some time as other things are higher priority (HA integration and COP calculation).
> 
> But why would you prefer the backup heater over the more efficient heat pump?

good question, i have also thought about this: i have considered that it might be more gentle for the compressor, especially in the summer months, if it does not only start briefly for the hot water heating.

in my case, the hot water is heated to 45 degrees per default, which takes approx. 1h. in case of pv surplus i switch the smart grid contact, the hot water is then heated up to 55 degrees. this again takes about 1h. this would mean 2 compressor starts per day, which could be saved by using the heating rod. out of the belly, the heating rod is wear-free, but the compressor is not. maybe the compressor will last a few years longer this way?! ;)

bausi2k, 2023-09-21:
As far as I know, you can only tell the heatpump to do it's work with a defined amount of power in kWh. 
Options are 
* Electrical Heater [TREU/FALSE]
* Room Buffering [TREU/FALSE]

Arnold-n, 2023-09-21:
> good question, i have also thought about this: i have considered that it might be more gentle for the compressor, especially in the summer months, if it does not only start briefly for the hot water [...]

True, though 30 starts/day is not uncommon in winter due to defrosts, so I would personally not worry about 2 starts/day for DHW.

> oh and with pv surplus the efficiency is rather secondary I think.

I guess that depends on your personal goals and local electricity pricing.

nicx, 2023-09-21:
> True, though 30 starts/day is not uncommon in winter due to defrosts, so I would personally not worry about 2 starts/day for DHW.

I will monitor that in the upcoming winter which is my first one ;) My Goal would be to have as minimum compressor starts as possible. so in best case its 1 and the heat pump ist just running all day (for heating my house).

bausi2k, 2023-09-21:
With the BRP06(whatever) you have full control about when and how long your heatpump runs, so you can optimise that. Defrosting is another issue of course… 

My heatpump is controlled by Loxone, because ist is much easier and also more energy efficient. I don't see any reason why my hp should run at 3am in autumn at moderate temperature…

nicx, 2023-09-21:
@Arnold-n another thought: i could switch the heating rod on and off more often, and depending on the pv surplus, i could also let it run for a short time. i could imagine that the compressor prefers it to run for a minimum time at startup and not to clock. it wouldn't matter to the heating rod, which could also be switched on every 5 minutes, for example.

Th301, 2024-01-09:
Hi @bausi2k ,
I found your comment here, that you are using the LAN-adapter for Daikin with Loxone.
I also use Loxone and I am still looking for the best possibility to read/write stuff in my Altherma 3 H/HT (with the old Rotex-Controll Unit).
Which heatpump and controller do you have? Do you have some additional info on your general setup between heatpump and Loxone? Would be cool! :)

bausi2k, 2024-01-10:
> Hi @bausi2k , I found your comment here, that you are using the LAN-adapter for Daikin with Loxone. I also use Loxone and I am still looking for the best possibility to read/write stuff in my [...]

@th301 
Hi, since Loxone doesn't know anything about ws(websockets) I'm using NodeRed to translate. 
I use the *Level 3* intergration of [the Loxone climate control](https://www.loxone.com/enen/kb/overview-indoor-climate/) where the Daikin EHBH/X(whatever) operations modes:

* switched on/off
* heating/cooling
* VLT/incomming water temperature
* tank operation on/oof
* tank water temperature
are controlled. 

for details on NodeRed Integration find this [thread & post](https://community.openenergymonitor.org/t/hack-my-heat-pump-and-publish-data-onto-emoncms/2551/45)

joncage, 2024-05-09:
Any updates on this @Arnold-n? Anything a fellow dev could help with?

I've got ESPAltherma hooked up to my EHVH11SU26CB6W but what I really want to do is do whatever it is the LAN adapter does to handle a PWM signal to adjust the power usage of the heatpump.

We have a 5kW cap on our export and on sunnier days are regularly hitting that so would be nice to have a gentle way to soak up the extra head into our DHW.

Since the LAN adapter supposedly can do this via PWM and my heatpump is apparently compatible with the BRP, I'm hoping there could be a way to do the same thing via your little module when hooked up to my RPI5 home assistant :)

Arnold-n, 2024-05-18:
Still on my activity list, for now the only power limiting option is to use the quiet levels which also limit the maximum power consumption.

Staacks, 2024-05-19:
Although I only did a quick manual test so far, I would like to add my thoughts about the differences between quiet mode and power limiting as I am not only interested in this because of solar power management, but also to improve efficiency.

If my heatpump draws full 2kW for water heating I observe a massive drop of the average heat exchanger temperature on the outside unit. This is of course not surprising, but it also reduces the COP as the unit has to work against a larger temperature difference. I expect that in a situation when time is not critical, slowly heating the water at 1kW or less would lead to a higher COP when the heat exchanger is not cooled down that much and can maintain a higher temperature.

In the release notes for v0.9.46 you mention that the quiet mode actually has a better efficiency then the default mode, which you attribute to lower power consumption of the fan. I suspect that it is actually due to the heat exchanger not being cooled down so much and the reduced fan speed would actually be counter productive here. So, if time is not an issue and if you are not concerned about noise, running the fan at max, but heating only at 50% (i.e. power limiting instead of quiet mode) might actually be even more efficient.

(Disclaimer: These are the thoughts of a physicists with a Carnot process in mind. Some insights by an engineer or just some experience from someone who tested this more systematically are very welcome :) )

Arnold-n, 2024-05-19:
Your experience matches with mine: that is why I prefer 24-hour operation at low load and also prefer to have 24-hour averaging of outside temperature to avoid having a higher load at lower outside temperatures when COP is worse.

But surprisingly, if the system is already in low-load, it still helps to reduce the fan setting. At 1kW power consumption in my 8kW system, silent level 3 does not limit the compressor any further but does limit the fan, giving a 5% higher COP:
[image]
[image]

Staacks, 2024-05-19:
Fantastic info, thanks. Can't wait to find some time (and good weather conditions) to compare with my system. Have been thinking about using 24h average as well. (But then we would be leaving the original topic of this issue.)

nicx, 2024-06-04:
> @nicx this may take some time as other things are higher priority (HA integration and COP calculation).

@Arnold-n many thanks for the HA and COP integration! :) Now I think we can go back to the origin topic question: Is the smart grid control still on your list? any idea when it will be implemented.

This feature would be the last missing feature to completely get rid of my ESPaltherma which I have running in parallel to P1P2MQTT ;)

Arnold-n, 2024-06-04:
You're welcome, thanks! Smart grid release is still (and high) on the agenda, but cannot predict yet when.

forTT78, 2024-09-28:
>  Smart grid release is still (and high) on the agenda, but cannot predict yet when.

Hi together, is there an update on this? :innocent:

Arnold-n, 2024-09-29:
Hope to provide this in October!

buennerbernd, 2024-11-04:
> Your experience matches with mine: that is why I prefer 24-hour operation at low load and also prefer to have 24-hour averaging of outside temperature to avoid having a higher load at lower outside temperatures when COP is worse.

Has your heatpump a setting to support 24-hour averaging of outside temperature?
I don't find this in my Altherma 3HHT. I would love to use such a setting. My heatpump reacts a little bit too fast on changing temperatures. So I have these two effects:

- I have an unnecessary overheating when there is just a single cold night. 
- If it was longer cold and the walls are still cold and it is the first warmer day at the outside, it is getting fresh in the house.

donjonsn, 2024-11-06:
Out of curiosity, will it be possible to support the same settings which are possible with the Dakin Home Hub EKRHH? The device has an extensive set of registers accessible by modbus. It looks like this is what is needed for this ticket, even better than the BRP069A61.

Arnold-n, 2024-11-06:
> > Has your heatpump a setting to support 24-hour averaging of outside temperature?
> 
> See e.g. 9.B.3 [1-0A]. But I don't know if this average value is visible via P1P2MQTT.

As far as I know the average value is not communicated over P1/P2. If the outside temperature is within the weather-dependent range points, it can be calculated from the current LWT setpoint. I plan to add temperature averaging in the code anyway.

Arnold-n, 2024-11-06:
> Out of curiosity, will it be possible to support the same settings which are possible with the Dakin Home Hub EKRHH? The device has an extensive set of registers accessible by modbus. It looks like this is what is needed for this ticket, even better than the BRP069A61.

Plan is to something similar indeed. It seems the EKRHH implements SG more direct, if I understand correctly it adds a dynamic power limitation via modbus registers 57/58, whereas the BRP069A61 calculates from the E_meter input what the power limitation should be.  However availability of the EKRHH is still very limited.

donjonsn, 2024-11-07:
> Plan is to something similar indeed. It seems the EKRHH implements SG more direct, if I understand correctly it adds a dynamic power limitation via modbus registers 57/58, whereas the BRP069A61 [...]

In addition, register 1 (inlet temperature) looks very promising to me because it allows to dynamically limit the heating to what is needed or do rooms heat buffering even if you do not have a Daikin room thermostat.

Arnold-n, 2024-11-07:
> In addition, register 1 (inlet temperature) looks very promising to me because it allows to dynamically limit the heating to what is needed or do rooms heat buffering even if you do not have a Daikin room thermostat.

Depending on the operation mode (LWT absolute or weather-dependent) it is already possible from HA to control power indirectly via the LWT setpoint resp. LWT deviation setpoint. The power is then determined by flow and difference with return temperature; the latter may be influenced by other factors depending on your heating system.

MichaelST01, 2024-11-07:
Does your Modul work with EKRHH? I thought it is not working? I have EKRHH working (Modbus-Mode), if you need to try something I would be happy to help.

Arnold-n, 2024-11-07:
Thanks! The bridge does not communicate with EKRHH, so EKRHH and the P1P2MQTT bridge will both act as auxiliary controller. I expect that when LWT is set from either that the last value will remain active. Would be interesting to see how dynamic power limitation is communicated in the P1/P2 traffic, so a log of P1P2/R# while hex data output is activated ('J3') and power limitations are changed over Modbus would be very useful.

MichaelST01, 2024-11-09:
I'll try to do it. I need to update my Modbus settings first (I actually don't use power limitations at all).
For installing of the EKRHH the firmware of the heatpump had to be updated. 
Also the modbus registers changed compared to MB/IO modul.

When I telnet to the bridge I get following msgs:
`[ESP] 2024-11-09 22:10:08 Unknown packetSrc 0x40 packetType 0x31 index 3
[ESP] 2024-11-09 22:10:08 Unknown packetSrc 0x40 packetType 0x31 index 4
[ESP] 2024-11-09 22:10:09 Unknown packetSrc 0x00 packetType 0x31 index 0
[ESP] 2024-11-09 22:10:09 Unknown packetSrc 0x00 packetType 0x31 index 3
[ESP] 2024-11-09 22:10:09 Unknown packetSrc 0x00 packetType 0x31 index 4
[ESP] 2024-11-09 22:10:09 Unknown packetSrc 0x00 packetType 0x31 index 11`

any hints how I can get rid of this? 

These are the modbus registers related to that:
[image]
Do you mean setting that? For dynamic power a special power meter is needed which I don't have.

Arnold-n, 2024-11-09:
The msg suggests the EKRHH gets another ID than 0xF0/0xF1/0xFF. I need to modify the code a bit to get rid of these but I would need a bit more info, can you either run [v0.9.56rc1](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.56rc1-Daikin-E.ino.bin) and look at the msgs again, or have a look at the hex data packet 00xx31/40xx31 where xx deviates from 0xF0/0xF1/0xFF which triggers these messages?

As for the power meter, I think it is one of the ways the EKRHH can do power limitation, but instead of a meter input signal I think you can also write the power limit directly to the modbus register (58 or 1002, not sure which).

MichaelST01, 2024-11-09:
I tried v0.9.56rc1. It is still the same msg. In attachment there is the hex log.
[Log unknown packetsrc.txt](https://github.com/user-attachments/files/17689564/Log.unknown.packetsrc.txt)

Arnold-n, 2024-11-09:
Thanks! The msg is slightly more informative, and the hex log is even more useful. The EKRHH takes auxiliary controller address 0xF2. I'll provide you with an updated firmware tomorrow that can keep history of the 0x31/0x32 messages.

Arnold-n, 2024-11-10:
[P1P2MQTT-bridge v0.9.56rc2](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.56rc2-Daikin-E.ino.bin) gets rid of the error messages and can inform you about changes in the 0x31 and in the new 0x32 packet from/to the EKRHH in topics like `P1P2/P/P1P2MQTT/bridge0/U/E/PacketSrc_0x40_Type_0x32_Byte_15` if you use the 'J102' command.

MichaelST01, 2024-11-10:
Hi Arnold, it works now.
Here is the log file where I changed General Power Limit (Modbus 58) to 19, 18, 17, 18, 19, 20  (kW)
If this is useful, I can try it with the specific Power Limit (Modbus 57) and the Operating-Modus (SG)

[log p1p2 modbus 58 (general power limit).txt](https://github.com/user-attachments/files/17693118/log.p1p2.modbus.58.general.power.limit.txt)

MichaelST01, 2024-11-25:
> [P1P2MQTT-bridge v0.9.56rc2](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.56rc2-Daikin-E.ino.bin) gets rid of the error messages and can inform you about [...]

There is no change in this values:
[image]

On the heatpump it was changes because the modbus register shows the new values.
[image]

Still getting some errors:
[image]

Arnold-n, 2024-11-27:
Many thanks! Seems packet 0x32 is surprisingly meaningless.

The EKRHH seems to use (previously unseen) parameter numbers 0x0093-0x00A0 and 0x0194-0x0195 in packet type 35, but these are mostly 0 all the time. The only parameter which matches your system changes is 0x0094 which seems to communicate the general power limit (with 0.5kW resolution, in u16div2 notation).

New P1P2MQTT-bridge firmware v0.9.57 should resolve all errors above and will output various SG related entities in subdevice HC_SG, perhaps one of these will show up to change during your experiments.

MichaelST01, 2024-11-27:
Hi Arnold, I installed v0.9.57 but in HomeAssistant I don't get values (even if I change) (I changed SG Mode Value and And Limit for "Empfohlen Ein")
[image]

when I changed "General Power Limit" new sensors appeared:
[image]

MichaelST01, 2024-11-27:
Long Log:
[log p1p1 modbus SG Mode new.txt](https://github.com/user-attachments/files/17937102/log.p1p1.modbus.SG.Mode.new.txt)

From that log I found out that this: repeats 6 times and is related to SG Mode: 3
00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
```
        Line   181: [MON] 2024-11-27 15:24:57 R T  0.025: 00F235950101FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
	Line   700: [MON] 2024-11-27 15:25:30 R T  0.025: 00F235950101FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
	Line 51082: [MON] 2024-11-27 16:20:12 R T  0.025: 00F235950101FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
	Line 51567: [MON] 2024-11-27 16:20:42 R T  0.025: 00F235950101FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
	Line 52362: [MON] 2024-11-27 16:21:32 R T  0.025: 00F235950101FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
	Line 52692: [MON] 2024-11-27 16:21:52 R T  0.025: 00F235950101FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE
```

this log should be SG Mode 2 (but not 100% sure)
00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
`	Line   381: [MON] 2024-11-27 15:25:10 R T  0.025: 00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
	Line   865: [MON] 2024-11-27 15:25:39 R T  0.025: 00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
	Line 51278: [MON] 2024-11-27 16:20:24 R T  0.025: 00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
	Line 51730: [MON] 2024-11-27 16:20:52 R T  0.025: 00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
	Line 52529: [MON] 2024-11-27 16:21:43 R T  0.025: 00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27
	Line 52853: [MON] 2024-11-27 16:22:02 R T  0.025: 00F235950100FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF27`

log related to SG Mode 0 not found

the times are related to the actual changes with modbus (I made the log larger to filter out other stuff)

I changed in total 6 times in the order SG Mode 3, SG Mode 2, SG Mode 0

Arnold-n, 2024-11-27:
I just uploaded v0.9.58rc2 which should now really fix the errors/unavailability of the SG entities and should make analysis easier... I have not yet renamed the info above to make a SG-mode entity, it is still called SG_Q_35-0195. In the messages above, the message 00F235 9501 01 encodes in packet type 35 that the single-byte parameter nr 0195 has changed to 01.

MichaelST01, 2024-12-11:
Hi Arnold, still no values at in SG section: Just HC_SG SG_General_Power_Limit shows "127" which is not correct.
Could this single Byte parameter Nr.195 be anyhow related with SG Mode? Is this somehow connected to : parameter list?
[image]

nicx, 2024-12-29:
I just installed 0.9.58 and got the SC sensors. Am I right that this is the current dev state and I cannot set the SG ready state yet?

nicx, 2025-01-08:
@Arnold-n could you perhaps give me a very short yes/no answer? :)

Arnold-n, 2025-01-08:
Sorry, slipped through, indeed currently SG control paramters can only be set via the cryptic E command and not yet from HA, for example setting limit parameter 94 in packet 35 to 19kW can be done with E35009426 (0x26=38, 38/2=19).

I think "127" means "no limit" - to be verified.

Confusingly there were 2 entities with the same name `SG_General_Power_Limit`, these are now renamed to `SG_35_0094-General_Power_Limit_kW` and `SG_General_Power_Limit_2`. Entity `SG_Q_35-0195` is (for) now renamed to `SG_35-0195-Mode3` in v0.9.58rc9.

MichaelST01, 2025-01-09:
[image]

This one works:
SG_General_Power_Limit_2 -> corresponds to Modbus 58 (Allgemeine Leistungsbegrenzung) 0-20kW (20 -> 127)

Changes in Modbus 57 (Leistungsbegrenz während Empfohlen Ein / Pufferung)
and
Changes in Modbus 56 (SG_Mode)

are not displayed in HC_SG

Arnold-n, 2025-01-13:
Thanks, I'll try to look into the others.

nicx, 2025-01-28:
Some more insights from my setup, maybe it helps in getting this smart grid control done :)

[image]

[image]

Arnold-n, 2025-01-28:
This suggests that 5 is not a power limit but a status?

nicx, 2025-01-29:
@Arnold-n no I don't think so. As far as I understand the sensor value "HC_SG SG_35_0093-General_Power_Limit_Active_Q" is directly reflecting the Smart Grid mode (free running = 0; recommended on = 1). after smart grid mode is switched tn "recommended on", the Daikin itself decides when to start extra heating. That's why the value "5" is set a little bit later in the sensors "HC_SG SG_35_0094-General_Power_Limit_kW" and "HC_SG SG_General_Power_Limit_ 2". This is exactly the time when the Daikin decides to start the extra heating.

savage7, 2025-03-04:
@Arnold-n I is there a chance that this work via the P1P2 adapter? 
If no I'll try to connect a shelly uni to control the Smart Grid option.

buennerbernd, 2025-04-01:
The only SG related topic I can see is SG_General_Power_Limit_2 (always = 127)
Are there any preconditions to get the other mentioned values? (Any configurations?)

I'm using pur MQTT, not HA.
I don't have any SG hardware connected to the heatpump.

buennerbernd, 2025-04-02:
I will answer my latest question by myself.
The other SG topics (0093 - 00A0) occurred immediately after I have enabled the controller mode by
`L1`

MichaelST01, 2025-04-04:
[image]
If I set Smartgrid-Mode to 3 (per Modbus) the value in the picture was changed to 3 (only if P1P2MQTT is in L1 controller mode, if it is L0 than values are unavailable)

nicx, 2025-04-04:
@MichaelST01 I am missing that sensor in HA. Where did you find it?

JohnVerbruggen, 2025-04-04:
Hello, Today I ugraded to 0.9.58rc19. Somehow I only see 'HC_SG SG_General_Power_Limit_2' in HA and none of the others mentioned above like @nicx .

MichaelST01, 2025-04-09:
> [@MichaelST01](https://github.com/MichaelST01) I am missing that sensor in HA. Where did you find it?

It is shown in MQTT / HC_SG / 
I installed Firmware: 0.9.58rc19
If Bridge is not in L1 Mode the value is "unavailable"

[4 more replies]