Arnold-n/P1P2MQTT #77: I have a sample set extracted, now how do I interpret the payload

BradleyFord, 2023-08-12:
I have managed to get a large data dump from my Daikin VAM; I have get this into Excel.

I can see various notes about different packet types, etc.

How do I take the next step to interpret this info?
(I am hoping to give Arnold an XLS with the various data interpted quite well, as I have about 6 or 7 different types of indoor units and VAM)

[image]

Arnold-n, 2023-08-13:
Hi @BradleyFord, many thanks for the logs.

Actually the amount of data of this model it not overwhelming; my analysis follows below.

Next step would be to add the 40, 41, and 42 packets to the ESP decoding firmware so you can monitor and try to understand any changing data in relation to your system's operation. I will ping you here when it is ready.

Your log shows the following packets:
-request 000015, always the same content "10 00   00 F0 FF FF    00 F0 FF FF    00 F0 FF FF    00 F0 FF FF"
-reply 4000FF ("not supported?")

-request 00001F, empty payload
-reply 4000FF ("not supported?")

-request 000040, payload length 20, no variation in your log, likely basic settings (on/off etc)
-reply 400040, payload length 19,  no variation in your log

-request 000041, payload length 15, only payload byte 6 changes; byte 5+6 may represent a temperature
-reply 400041, payload length 9,  only byte 5+6 change, may represent a temperature

-request 000042, payload length 8, only payload byte 6 changes
-reply 400042, payload length 11, only payload byte 5 changes

-request 000020, payload length 1, no variation, sensor request?
-reply 400020, payload length 20, no variation

-requests 000120 .. 001020, payload length 1, no variation
-no reply

-requests 00F030, payload length 20
-no reply (no auxiliary controller present)

-request 0000A1, payload length 16, all-zeroes
-reply 4000FF ("not supported")

-request 0000B1, payload length 18, all-zeroes
-reply 4000B1, payload length 18, all-zeroes

-request 800018, payload length 11, no variation
-no reply

-request 808000, empty payload
-no reply

BradleyFord, 2023-08-14:
Great thanks for that, although it would be good if I could help to interpret the payload for you!!!

I will be back home again on Sunday, so can upload the changes then.

It seems I temporarily broke the unit for the other VAM. When configuring I added the wifi, but didnt configure the MQTT and could not find the WebUI to add the MQTT info after configuring Wifi.
So I will reflash it at the same time as well back to defaults with these enhancements, then set Wifi and MQTT at the same time.

The bigger one will be the AC units, that shall be fun :), although that will take me some more time to physically install.

Arnold-n, 2023-08-14:
When the data is reported over MQTT you can definitely help to interpret!

A webUI for MQTT reconfiguration is planned. For now, if you restart (with the reset button) the unit while WiFi is unavailable, it restarts the webUI to configure WiFi and MQTT. Alternatively, you can telnet to it and enter MQTT settings with the B command (B MQTT-IPv4 [MQTT-port [MQTT-user [MQTT-passwrod]]]). Reflashing does not reset MQTT settings unless you explicitly erase all flash memory before doing so.

BradleyFord, 2023-08-21:
I just uploaded another batch of data, also I recorded a video as I slowly go through adjusting each of the settings.
[VAM Raw Data r2.xlsx](https://github.com/Arnold-n/P1P2Serial/files/12395141/VAM.Raw.Data.r2.xlsx)

Here is a link to the video:
https://photos.app.goo.gl/Z3VgmSTmc2sqPWWd9

BradleyFord, 2023-09-21:
Hi Arnold,
Is there anything I can do to help with this?

linhpd88, 2026-02-03:
> I ended up purchasing a Daikin BACNET interface

Thank your for the information.