Arnold-n/P1P2MQTT #81: Usage without existing Daikin control module

rubenvanparys, 2023-09-09:
I was wondering, can the presented module work without a Daikin control module attached to the P1P2 port? I have an Altherma 3 without central module but with external thermostat that just gives an on/off relay signal on a different port. 

If so, can the P1P2toMQTT do the following things:
- Turning on/off room heating/cooling
- Set dynamically the current/power limit
- Read the on/off state of the external thermostat
These are all things that can be set/read from the heathpumps display, but not sure if they are accessible over P1P2?

Arnold-n, 2023-09-10:
Hi @rubenvanparys ,
Most newer systems have an LCD controller in the inside unit which acts as main controller so even if the P1/P2 port seems empty, any new device should and can still act as auxiliary controller.
At the moemnt you can turn heating/cooling on/off. I haven't looked into the on/off state of an external (relay) thermostat, but I think that is very likely possible. Current/power limit setting should be(come) possible too but is not supported yet in the current version v0.9.41.

Arnold-n, 2024-05-18:
Updates:
-P1/P2 communicates the status of the heating/cooling relay inputs for both main and additional zones, regardless of the mode (room thermostat, external thermostat, or LWT), so the P1P2MQTT bridge outputs these too (starting at v0.9.48) so they can be used as input in HA.
-Power limit setting via P1/P2 is now available via the quiet level selection, which can limit maximum power consumption to a few fixed levels.

closed 2024-06-05 by Arnold-n