Arnold-n/P1P2MQTT #83: Simulate Madoka room thermostat with P1P2Serial

nicx, 2023-10-20:
Is it possible to simulate an external Madoka room thermostat with the P1P2serial module?

My idea: activate the external room thermostat function in my Altherma, then setting the current room temperature with the p1p2 module by writing it via MQTT to the Altherma. with that, I could set the desired room temperature and control the behavior of the Altherma.

My understanding is that the only missing feature is writing the room temperature to the correct register.... or maybe this is already possible?

Arnold-n, 2023-10-20:
This is not yet possible (or not yet known how to), but I hope to dive into this and get it to work. If not, I have been working on a work-around to achieve the same effect.

Arnold-n, 2024-05-18:
A fake room temperature can now be generated by the P1P2MQTT bridge for Daikin E-series systems. This may help to avoid an error message on certain systems that the bridge does not generate a temperature, but we do not know which systems require or accept this.

Command `F1` will switch this function on, with default reported room temperature of 20.0C. Command `F1 205` will change the fake temperature sensor to 20.5C. The permitted room temperature range is 0.0C-25.5C. Command `F0` will switch this function off. By default this function is off. It will also switch off after an ATmega restart, the state is not yet stored in EEPROM.

DiabloVmax1200, 2024-07-14:
Hi I tried this option today with the v0.9.53 two things I mentioned.

- The "F1" is reset after a restart of the Heatpump (as you have written above) Could you store it in the EEPROM?
- at my Daikin Altherma 3 GEO - (EGSAX06DA9W) the Room temperature is always 25.6°C I can't change the value to a different one. this value is shown in MQTT and on the display

Arnold-n, 2024-07-15:
Can you test P1P2Monitor v0.9.54? It should store the setting in EEPROM and should fix the 25.6C bug.

adam-the, 2024-09-29:
Hi Arnold. I've just setup my HA to send the current room temperature to the P1P2 bridge on change, but it seems as though the value isn't actually getting faked, and hence the heating is staying on permanently. Unless I'm misunderstanding how this feature is meant to work?
I don't have the thermostat plugged in, only the main controller, but even with the thermostat in, the bridge's fake temperature value wasn't being respected.
<details>
<summary>Screenshots</summary>

[image]
[image]

</details>

Arnold-n, 2024-10-05:
Sorry, I am not sure how this function works in the Madoka. The temperature sensor simulation was originally added to avoid an error message message on the Madoka - perhaps there is a configuration option to select a 2nd/external sensor?

closed 2025-03-31 by nicx