Arnold-n/P1P2MQTT #84: P1P2 Serial will not connect to Wifif

PietSmits, 2023-10-22:
Yesterday I had to reset my router and since then P1P2 Serial has lost connection. When I connect to the AP and input the credentials it does not connect anymore. I have not had this issue previously, it always reconnected without issue, even when its power was switched off/on.
After entering the credentials and saving it just opens the AP again. Unfortunately I cannot find where things go wrong.

Any suggestions to make the module connect again?

Arnold-n, 2023-10-22:
@PietSmits, this may be difficult to debug. It sounds like a poor WiFi signal, or wrong SSID or password, or a router which no longer accepts this ESP as client. You didn't change the SSID or password or other settings on your router?

The only way to debug this is to listen-in on the serial output of the ESP after you set the credentials, how to do that (with the extra ESP-01 that was included with your interface) is described in Annex 2 of your installation guide. Please let me know if you need any help with this.

PietSmits, 2023-10-23:
Thanks Arnold, although all my other ESP's connected without any issues I decided to set up the 2.4GHz bit of my router from scratch. After I did that I did see that the module connected, but lost connection again after a few minutes. I decided to leave it for the night, and lo and behold this morning all data was coming in again. 
No idea what happened but it seems to have solved itself for now.
Thanks for your input, it triggered me to do the full reset.

Arnold-n, 2023-10-23:
Thanks Piet, let me know if the problem returns. One known cause for "lost connection after a few minutes" is if the MQTT connection isn't restored, but even then it should reconnect to WiFi to retry.

closed 2023-10-23 by Arnold-n

gmcmicken, 2023-10-23:
I had a ton of trouble with my Wi-Fi and the P1P2 esp, it would sometimes connect and other times not, or take hours. Eventually I figured it "airtime fairness" of my Wi-Fi router in advanced wireless settings needed to be turned off.

Arnold-n, 2023-10-23:
Interesting, thanks for that hint! In that case it might help to change the j-mask setting: if you change it from 13003 (default) to 3002 a lot less data is being communicated as the raw hex data and non-HA parameters are no longer communicated. This can be done with the command "J3002" via telnet or via topic P1P2/W.

gmcmicken, 2023-10-23:
> Interesting, thanks for that hint! In that case it might help to change the j-mask setting: if you change it from 13003 (default) to 3002 a lot less data is being communicated as the raw hex data [...]

The issue only manifested in a delay to connect not actually in the quality once connected... strangely! I use output mode J2 so there's not a whole lot of traffic.