Arnold-n/P1P2MQTT #85: FXMQ-P7 Series

Enelar, 2023-11-10:
I will use this thread to organize my research process on [FXMQ-P7 series] (http://www.daikintech.co.uk/Data/VRV-Indoor/FXMQ-P/2014/FXMQ-P7VEB/FXMQ-P7VEB_Databook.pdf)

My indoor unit is FBQ60C7VEB, my outdoor unit is RXS60F2V1B.

There is an protocol dump where I pressed turn on and off button at 24:00 and 24:10
https://docs.google.com/spreadsheets/d/1w95OQW-eVxMFEjkyJB_ZmwWfgBITxUhmqZfW2W5A39E/edit?usp=sharing

It seems that protocol is different from https://github.com/budulinek/arduino-altherma-controller/blob/main/Payload-data-write.md

gmcmicken, 2023-11-10:
> I will use this thread to organize my research process on [FXMQ-P7 series] (http://www.daikintech.co.uk/Data/VRV-Indoor/FXMQ-P/2014/FXMQ-P7VEB/FXMQ-P7VEB_Databook.pdf)
>
> My indoor unit is FBQ60C7VEB, [...]

Should work fine with the existing F series FXMQ code in this library (not the altherma code)

Arnold-n, 2023-11-10:
Hi @Enelar, your bridge is flashed with the generic F-series, so you could first try with "L5" to see which 00F03x messages it triggers. If your system is indeed FXMQ-like (I had not derived that from the FBQ product code), you could flash the ATmega with the  the FXMQ code as @gmcmicken also mentions.

Arnold-n, 2023-11-10:
Hi @Enelar, your bridge was originally flashed with the Altherma (E-series) code, as I wasn't aware that you have a FBQ system and assumed you have an Altherma system. So it would be good to flash the FXMQ code on the ATmega and the F-series code on the ESP. Afterwards I would recommend to start with "L5" to see if the 00F03x messages it triggers match the FXMQ structure before trying "L1" mode.

Enelar, 2023-11-12:
Thanks a lot for your support. However I'm still experience issues. Following the guide I'm in L1 mode.
And during power on-off cycle I intercepted this packets:

```
P1P2/R/033 R 2023-11-12_02:27:53 00F03800006101180011001A001101002A00C0E4
P1P2/R/033 R 2023-11-12_02:27:54 00F03803006101180011001A001101002A00C03C
P1P2/R/033 R 2023-11-12_02:27:54 00F03801006101180011001A001101002A00C03D
P1P2/R/033 R 2023-11-12_02:27:54 00F03801006101180011001A001101002A00C03D
P1P2/R/033 R 2023-11-12_02:27:55 00F03801006101180011001A001101002A00C03D
P1P2/R/033 R 2023-11-12_02:27:55 00F03802006101180011001A001101002A00C0E5
P1P2/R/033 R 2023-11-12_02:27:55 00F03800006101180011001A001101002A00C0E4
P1P2/R/033 R 2023-11-12_02:27:55 00F03800006101180011001A001101002A00C0E4
```

Which follows this documentation perfectly: https://github.com/Arnold-n/P1P2Serial/blob/main/LogicalFormat/Daikin-protocol-FDY125LV1.md

I tried to reply this data
```
P1P2/S/033 * [ESP] To ATmega: ->F 3803006101180011001A001101002A00C0<-
P1P2/S/033 * [MON] 2023-11-12_02:34:55 wr_nr invalid, should be 0, 1, 2, 4, 6, 8 (or for packet type 0x3B: 16 or 17): 3
P1P2/S/033 * [ESP] Uptime 1400 at 20231112_02:34:59
P1P2/S/033 * [ESP] To ATmega: ->F 3801006101180011001A001101002A00C0<-
P1P2/S/033 * [MON]:w 2023-11-12_02:35:04r_val for payload byte 1 (operating-mode) must be in range 0x60-0x67
```

It seems that it does not like 03 mode (turn on), and with 01 mode it doesn't expect second byte (00).

I tried without it
```
P1P2/S/033 * [ESP] To ATmega: ->F 38036101180011001A001101002A00C0<-
P1P2/S/033 * [MON] 2023-11-12_02:35:41 wr_nr invalid, should be 0, 1, 2, 4, 6, 8 (or for packet type 0x3B: 16 or 17): 3
P1P2/S/033 * [ESP] Uptime 1450 at 20231112_02:35:49
P1P2/S/033 * [ESP] To ATmega: ->F 38016101180011001A001101002A00C0<-
P1P2/S/033 * [MON] 2023-11-12_02:35:53 Initiating write for packet-type 0x38 payload byte 1 to value 0x61
```
but with no reaction of AC.

I wasn't able to find any documentation on FXMQ series protocol so I kinda suck here
https://github.com/Arnold-n/P1P2Serial/tree/main/LogicalFormat

Enelar, 2023-11-12:
Update: success with
```
F 38 01 61
F 38 01 62
F 38 01 61
F 38 00 01
F 38 00 00
```
Now it makes sense, instead of providing whole 16 byte string I can alter only value I need to change. Very neat!

closed 2023-11-12 by Enelar

gmcmicken, 2023-11-12:
> Update: success with
> ```
> F 38 01 61
> F 38 01 62
> F 38 01 61
> F 38 00 01
> F 38 00 00
> ```
> Now it makes sense, instead of providing whole 16 byte string I can alter only value I need to change. Very neat!
> 
> 

Sorry, I don't know where you got the impression you needed to send a manual payload, the documentation specifies the commands as you've now found them and the library packages it up into a proper payload. :-)

Arnold-n, 2023-11-12:
Great to see your system works! I'll start documenting FXMQ protocol data, as indeed that is still missing, though your system mimics FDY. I have also added a separate file with the F-series commands. If you have any full logs in L1 mode or more experiences what works or not, I'd really appreciate it (via here or DM).

Enelar, 2023-11-12:
> Sorry, I don't know where you got the impression you needed to send a manual payload, the documentation specifies the commands as you've now found them and the library packages it up into a proper payload. :-)
I couldn't find any documentation on F series... I only found mention of F command that not yet supported... So I had to improvise.

Also it was very surprising to me to learn that my AC is not Atherma 3. I thought it's like generations of their motherboards, 1990-2010, 2010-2018, 2018-present time, or something.

> If you have any full logs in L1 mode or more experiences what works or not, I'd really appreciate it (via here or DM).
Sure, I can share both, will try to gather something usefull.

Also, since byte 4 and 8 only affect FAN speed, maybe lets set possible values to 0-2, instead of calculating hex numbers with no reason. (Like F 38 04 02 -> Max fan speed, L 38 04 00 -> Min fan speed)
https://github.com/Arnold-n/P1P2Serial/blob/main/LogicalFormat/Daikin-protocol-FDY125LV1.md

Arnold-n, 2023-11-12:
I think the name Altherma is used only for air-to-water and water-to-water heat pumps (models start with an E), not for AC units (starting with an F).

I like your suggestion to use lower fan speed numbers. I will have 0/1/2 translate to 0x11/0x31/0x51, but also keep the original possibility to use values in the 0x11-0x51 range. @gmcmicken, did you draw conclusions from experiments with other values than the default ones for fan speed?

gmcmicken, 2023-11-12:
> I think the name Altherma is used only for air-to-water and water-to-water heat pumps (models start with an E), not for AC units (starting with an F).
>
> I like your suggestion to use lower fan speed [...]

Yes you can set one value higher than the maximum x51 but I do forget what that value was exactly, and I don't think it needs to stay in there so you can do as you like.

gmcmicken, 2023-11-12:
I think it would be helpful to update the "F Series" readme with this nomenclature document. If we use "major version" as a compatibility starting point it might clear some things up.

[F Series Nomenclature.pdf](https://github.com/Arnold-n/P1P2Serial/files/13328955/F.Series.Nomenclature.pdf)

Arnold-n, 2023-11-12:
Thanks, it seems plausible that the first letter after the heat capacity is a generation, but for generation L there are still different data protocol variations, as so far we have 3 "major protocol variations" identified:
-FDY-LV1 seems similar to 2002 FBQ-B, ADEQ-B2VEB, FBQ-C7VEB, and perhaps FBQ-B7V1 (major versions: B, C, L)
-FDYQ-MV (major version: M)
-FXMQ-PWM seems similar to FXMQ-PAVE, FDYQN-LAV1, perhaps FBA-A9 (major versions: P, L, A)

gmcmicken, 2023-11-12:
> Thanks, it seems plausible that the first letter after the heat capacity is a generation, but for generation L there are still different data protocol variations, as so far we have 3 "major protocol [...]

Interesting. Perhaps the protocol might match closer to the manufacture year, or the controller model (for example my FTQ series uses BRC1E71 and is from 2012)

gmcmicken, 2023-11-23:
> Interesting. Perhaps the protocol might match closer to the manufacture year, or the controller model (for example my FTQ series uses BRC1E71 and is from 2012)

I'm not sure why I didn't think of this earlier as my model is a two letter major version (PB, and the field settings are quite different from the PA version). I think it's the same discrepancy as L and LA.