Arnold-n/P1P2MQTT #88: WiFi backdoor

Enelar, 2023-11-14:
This vector is highly unlikely, but still exists.
Each time device looses connection, it will reopen public hotspot.
The wifi password of this hotspot is already in databases.
If anyone would just drive by in a car, send wifi deauth beacons, it will make device expose itself.

My proposal: if the device has no wifi connection, create hotspot only after restart (physical key press)

Arnold-n, 2023-11-14:
You mean it is in databases other than github?

My concern is that the device does not only reset if the reset button is pressed, but also after a Daikin system restart (if powered via P1/P2), or when the known WiFi is no longer available to retry to connect (and if it fails, to start the AP). So how should we differentiate between reset-button-restarts and other restarts? Tasmota uses a method of repeated restarts, but this must be at the cost of flash writes, which may be a risk in some situations when the device continues to restart every 2-3 minutes.

Enelar, 2023-11-14:
Maybe if device has been restarted faster than 1 second after initial start?

Arnold-n, 2024-05-18:
@Enelar : Thanks for your suggestion, this is implemented in v0.9.46:
-the AP SSID and password can be configured by the user, and
-if the password is no longer known, a double-tap of the reset button makes an AP available with the default SSID/password. This mode is visible as the blue LED will remain be on in the 2nd case.
Not sure whether Tasmota uses flash - I doubt it, as it can be done via the RTC memory, which is what P1P2MQTT uses.

closed 2024-06-05 by Arnold-n