Arnold-n/P1P2MQTT #93: Control outdoor unit without requiring an indoor unit?

PeteC1773, 2023-12-18:
Could this project be used to switch on an outdoor unit without the use of anything else (IE an indoor unit, control module etc)? Ideally I would be able to make the outdoor unit run and control its output (what parameters are made available to the indoor unit I don't know -  for instance maybe the indoor unit controls the EEV or maybe it just asks for a particular amount of cooling / heating - but somehow the outdoor unit can ramp up and down its compressor power, vary the EEV opening and control the reversing valve, whether under its own control or the indoor units control IDK). 

Why:
I have access to free outdoor units sometimes (never with a matching indoor unit though) and I would like to hack together a water to water heat pump. I can make an open loop GSHP system as I have ground water. Buying a heat pump like this would be very expensive and I have experience with the brazing, charging etc of non inverter HVAC systems like these but since most of the units that are made available to me now through my local scrap yard are inverter, and since inverter is much better, I would like to use an inverter system to cannibalise. I would use a water heated evaporator of my own construction and an indoor underfloor heating loop driven via a plate heat exchanger - so I need to hack the original aircon unit completely to bits. Controlling the compressor myself is the sticking point though.

Thanks, Pete

Arnold-n, 2023-12-19:
Not at this moment, as at this moment the project only supports P1/P2, which is only used for the connection to the indoor unit on split systems (and to monoblocks outside).

PeteC1773, 2023-12-21:
OK thanks. Also thanks for getting back to me on that.

Kind regards
Pete

closed 2024-05-18 by Arnold-n