Arnold-n/P1P2MQTT #94: Write parameter via MQTT unreliable

nicx, 2024-01-12:
I use the Home Assistant to set the hot water temperature several times a day depending on the electricity price. For example, I publish the value "E 35 2F 1" in the topic P1P2/W to set the hot water temperature to 30 degrees.

Unfortunately, this rarely works... Strangely enough, it sometimes works if I simply publish the same value again without resetting the P1P2 module. Most of the time, however, it only works exactly once, then not again until the module is reset.

In my MQTT Broker the value is reliably published by HA.

It seems that either the P1P2 module only sometimes reads the value (why?), or the heat pump only sometimes accepts the value from the P1P2 module (why?).

Arnold-n, 2024-01-12:
This is intentional - P1P2Monitor does not allow more than 1 parameter write per hour to avoid too much flash wear: see [README](https://github.com/Arnold-n/P1P2Serial/blob/03f53dc752166231369eb8cdedb86b0f0fb4c85f/README.md?plain=1#L149)  and [this discussion](https://github.com/Arnold-n/P1P2Serial/discussions/89).

You should see an error message on P1P2/S/# that you are out of write budget.

In a next version I will make the limit flexible but allowing more writes is "at your own risk".

nicx, 2024-01-13:
@Arnold-n hm ok, understood. but in my case Ill do writes only 4 to 8 times a day, and there is a 3 hour pause to the next write. so it should work an do not harm any budget.

is there a counter or sensor regarding the write budget? in my case for example the A8_Writes_Refused sensor is always 0.

Arnold-n, 2024-01-13:
@nicx Can you check the P1P2/W/# topic for write instructions? In your log of less than 1 hour there are really 7 parameter writes (and perhaps more failed  attempts): 2 to switch heatpump on (35:2F -> 01/01) and 5 writes to change the DHW setpoint (36:03 -> 40.0/35.0/30.0/43.0/30).

You can make the write-budget visible by adding 0x20000 to the J-mask, and restarting mqtt-discovery for HA with D3.

Writes_Refused only counts writes which are refused due to command overruns (which will be solved by buffering); perhaps another counter is useful. There is a counter "Counter_Parameter_Writes" but by default it is not output unless you add the 0x20000 mask option.

nicx, 2024-01-14:
@Arnold-n you are right, these multiple writes where my manual tests... but that is not the normal case ;) i didnt now the budget-feature before.

could you give an answer to my second question? because maybe that is my problem: i am setting the write parameter in mqtt regularly, even if ther is no change. for example the on/off setting you can see in the log, the daikin is on, i am setting the write parameter to turn on. the parameter changes max twice a day. but it is set every hour.

Arnold-n, 2024-01-14:
The module always performs a parameter write action when asked to with the 'E' command. This could be filtered but so far I did not see the need for it.

nicx, 2024-01-14:
@Arnold-n Unfortunately, I don't quite understand: what do you mean by "asked for it"? 

example: i publish the value "E 35 2F 1" in the topic P1P2/W at 10:00 a.m. the same happens at 10:30 a.m., i.e. the identical value is published again. between 10 and 10:30 there is no publishing of any value.

do 2 write processes then take place? is the time of the publishing the write trigger and not, for example, a value change?

Arnold-n, 2024-01-16:
Indeed I meant that each publishing on P1P2/W, even if a duplicate of an earlier value, leads to a "parameter write" on the P1/P2 interface for the main controller; but as this is not a real change (unless there was a manul change in between) this does not lead to a real parameter change in the indoor unit. I think this can be filtered in the P1P2MQTT interface but I am not sure it will work in all circumstances. For now I would recommend to filter it at the source: in HA.

nicx, 2024-01-16:
I changed the automation in HA for now and will monitor the behaviour. Thanks so far @Arnold-n for clarifying.

closed 2024-06-05 by Arnold-n