Arnold-n/P1P2MQTT #98: Cannot set quiet mode on an Altherma 3 R W

Staacks, 2024-05-19:
Just installed to test the power limits in quite mode (as suggested in https://github.com/Arnold-n/P1P2MQTT/issues/69#issuecomment-2119007683), but I am not able to set the quiet mode / quiet level.

In HA I see Quiet_Level, Quiet_Level_When_On and Quiet_Mode. The latter do not reflect the actual setting of the heatpump and changing them has no effect. Quiet_Level actually changes correctly if I manually set the quiet level on the heatpump's control panel, but changing the entity does not have an effect and its value just resets to the old state.

Via telnet I can actually see that the HA entity results in an action on the P1P2bridge:
```
[MON] 2024-05-19 12:28:59 Initiating parameter write for packet-type 0x3A parameter nr 0x4D to value 0x1
[MON] 2024-05-19 12:28:59 Initiating parameter write for packet-type 0x3A parameter nr 0x4C to value 0x2
[MON] 2024-05-19 12:29:00 Executing E command for packet 0x3A setting param 0x4D to 0x1
[MON] 2024-05-19 12:29:00 Executing E command for packet 0x3A setting param 0x4C to 0x2
```
I am running an Altherma 3 R W heatpump (EHBX08EF9W) without any additional P1P2 devices (no wall panel, no thermostat, no LAN adapter etc.). (Arnold, you might remember that we were in touch via email in March.)

Also, there are a few things that surprised me about the quiet mode on my pump and I am not sure if that is normal or the reason that it is not working: I am not allowed to change the quiet mode on a user level (only with the installer code) and the quiet mode is not reflected by any field. Both are also documented in the manual (sorry, these are screenshots of the German manual, but hopefully it still illustrates what I mean).

[image]
The two asterisks mean "only accessible by the installer", so I am allowed to change the programming if in auto mode, but noch change the mode or level as a user.

[image]
This is the quiet mode level setting and the empty second column is where other settings have a field code associated with them.

Not sure if any of this is unusual or related with the problem.

Arnold-n, 2024-05-19:
German is fine for me, thanks for the information, and I do remember! Quiet level setting appears to be complicated and communication happens via different message formats (0x35 and 0x3A) and took some time to get working here. Your system is at least different in that it has a 4th "automatic" mode in 7.4.1, and that it may indeed depend on the authorization level. Perhaps the P1P2MQTT bridge must claim to be in installer level before changing quiet level, which would make things complicated.

What would be helpful is to log the 0x35 and 0x3A messages when you manually change your quiet level and when you change authorization level. This can be done by setting `J12B` or even better also log all hex data with `J13B`.

Staacks, 2024-05-19:
I was also surprised to find it represented as a single setting because in the menu it is represented as OFF, MANUAL and AUTOMATIC and you can only set three levels "LEISE", "LEISER" and "AM LEISESTEN" (German for quiet, quieter, quietest) in MANUAL mode.

In AUTOMATIC there are also menus to program times as well as an installer menu where you can set limitations for the user, so he can only set certain levels at some time intervals (these restrictions are all disabled at the moment).

Here are the logs with J13B. I recorded a few seconds each while switching from OFF to LEISE, from LEISE to LEISER, from LEISER to AM LEISESTEN and finally back from AM LEISESTEN to OFF:

[P1P2-quiet-mode-am-leisesten-to-off.log](https://github.com/Arnold-n/P1P2MQTT/files/15369391/P1P2-quiet-mode-am-leisesten-to-off.log)
[P1P2-quiet-mode-leiser-to-am-leisesten.log](https://github.com/Arnold-n/P1P2MQTT/files/15369392/P1P2-quiet-mode-leiser-to-am-leisesten.log)
[P1P2-quiet-mode-leise-to-leiser.log](https://github.com/Arnold-n/P1P2MQTT/files/15369393/P1P2-quiet-mode-leise-to-leiser.log)
[P1P2-quiet-mode-off-to-leise.log](https://github.com/Arnold-n/P1P2MQTT/files/15369394/P1P2-quiet-mode-off-to-leise.log)

Arnold-n, 2024-05-19:
Thanks, and also emailed you about this. It appears from your logs that the system does not communicate quiet settings via 0x35/0x3A packets from/to the P1P2MQTT bridge. This means that for now some or all Altherma 3 R systems (=? systems having an 'E' after the capacity number) may not support control of quiet mode and level setting, only monitoring.

forTT78, 2024-09-30:
Hi, it looks like I have a similar situation on my System:
Indoor: EHSXB08P50EF
Outdoor: ERGA06EAV3H

I can see the change in HA via telnet, but no change at all at the system. 
It looks as if only the installer is allowed to make a specification and the user is not.