Arnold-n/P1P2MQTT #99: MQTT settings lost during firmware update

Staacks, 2024-05-19:
I just updated both controllers to the latest firmwares (P1P2MQTT-bridge-v0.9.50-Daikin-E.ino.bin and P1P2Monitor-v0.9.46-Daikin-E.ino.hex via OTA, web and avrdude respectively).

After the update, the P1P2bridge tried to connect to an MQTT broker at 192.168.4.12 with default P1P2 user and password instead of the 192.168.2.x IP and the user and password I had set. The Wifi credentials were still the correct ones from before the update, so it was correctly connected to my network with a 192.168.2.x address.

This made setting new MQTT credentials via webinterface impossible (no 192.168.4.1), but I could set them via telnet, so everything is fine for me.

Note that I updated from a prerelease v0.9.45rc30 (Arnold, you sent me that one in March), so if this is only an issue with this test rc version, the issue can be closed immediately.

BTW: The documentation you sent me mentions a B command to set up MQTT via telnet. This command was not available and I used P instead.

Arnold-n, 2024-05-19:
Indeed that documentation is outdated, the B and I commands have been replaced by the more generic P command. I have tried to keep EEPROM (which includes MQTT settings) as much as possible from version to version but indeed sometimes this may have gone wrong especially for rc versions shared individually, sorry about that. And the only two options to recover from wrong MQTT settings is:
-telnet as you did, and use the commands `P7 IPv4-MQTT-server`, and if necessary also `P8 MQTT-port-nr`, `P9 MQTT-username` and `P10 MQTT-password`
-double-tap reset-button until blue LED lights up, and enter WiFi+MQTT credentials in AP (SSID/password `P1P2`/`P1P2P3P4`) via WiFiManager

Staacks, 2024-05-19:
No need to be sorry, your work is much appreciated. Just wanted to avoid others to get stranded. So, if you think it is indeed rc version specific, this can be closed.

Arnold-n, 2024-05-19:
I'll keep it open for a while for a few other rc users, and in case more people encounter it. I'll also add a few lines to the HA doc.