Arnold-n/P1P2MQTT #101: Sensors for Power Consumption and Realtime COP not working

nicx, 2024-06-04:
Maybe it's the same problem: My sensors for Power consumption (HC_Power) and Realtime COP (HC_COP) are always 0. Power Production Sensors are working, COP-Lifetime seems also correct. What could I check to find the reason for that? :)

Arnold-n, 2024-06-04:
It's the same problem: power consumption and real-time COP calculation rely on an external electricity meter providing data via `P1P2/P/meter/U/9/Electricity_Power`. Daikin unfortunately does not provide any real-time consumption data via the P1/P2 interface. So without external input, these values remain 0.

HomeWizard meter users may use a local API and a separate ESP8266 to request data from the meter and copy it to MQTT.

nicx, 2024-06-04:
@Arnold-n ah ok. understood. so could I populate thempower value from my espaltherma to this topic and then the COP is calculated automatically? should the value be the total power of the heatpump with or withour the backup heater?

Arnold-n, 2024-06-04:
Yes, but it may be less accurate than a real external electricity meter, as the current reported by ESPAltherma has low resolution (0.2-0.3A), and does not take blind power into account.
For COP the compressor power would be needed. The code for the backup heater power input is not complete yet (and less useful).

buuhsmead, 2024-07-18:
I am updating topic 'P1P2/P/meter/U/9/Electricity_Active_Power' with current used Watts of the WP.
And the topic 'P1P2/P/P1P2MQTT/bridge0/C/9/COP_Realtime' is not updated.
Is there a time differences or other conditions that should be filled before?

buuhsmead, 2024-07-19:
@Arnold-n The value in Watts is allowed to have a decimal point or should it be an int [0-32767].
Currently, I am sending 17.1 or 18 or 18.4.

Arnold-n, 2024-07-19:
The entity provided by `P1P2/P/meter/U/9/Electricity_Active_Power` should be repeated at least once/minute. But COP_Realtime is currently only calculated for heating, and not for cooling, I will add that in the next version.

The value is rounded down (actually it just ignores any character other than [0-9]).

adam-the, 2024-09-28:
Is there any potential harm in sending the value more frequently? i.e, every second or every 15 seconds?

Arnold-n, 2024-09-29:
No harm at all, and it increases accuracy of the real-time COP reporting. The real-time COP calculation is done almost once/second and takes the most recent input values for active power and for heat production (flow * delta-T) unless power information is too old (>1 minute). Active power is not (planned to be) integrated over time; for this Electricity_Total will be used.

And as I just corrected above and in the documentation, by default it is `P1P2/P/meter/U/9/Electricity_Power`.

hdaems, 2024-09-29:
Adjusting to P1P2/P/meter/U/9/Electricity_Power the COP is calculated, both realtime and lifetime.
For me this can be closed.

closed 2025-03-31 by nicx