Arnold-n/P1P2MQTT #102: Heating_Only is not correct for my system.

miguelangel-nubla, 2024-06-04:
I have a EHVX08S23DJ6V7 indoor unit which definitely has cooling mode reporting `Heating_Only=1`, therefore `HC_LWT Abs_Cooling` climate entity is unavailable.

Staacks, 2024-06-04:
On my system some of the field settings are communicated if I scroll through them on the heatpump's interface. I need to enter the installer's code, then go into the "Installer settings" and the entry "Overview field settings" at the bottom. Correct parameters including the availability of cooling become available on the bridge as soon as I scroll through the pages there. Unfortunately, this needs to be done each time the bridge is restarted (a heatpump restart is no problem), but the bridge pretty much never needs to be restarted once it is working properly.

Arnold-n, 2024-06-04:
@Staacks Thanks - but actually if the WiFi and the same MQTT server is immediately available after restart and retained data can be read back, and the bridge was not powered down, only restarted, it should maintain the heating-only and cooling-capability status via the MQTT-retained messages and via the ESP8266 RTC memory. Only if the RTC memory is invalidated (power-down, AP?), or if the MQTT server retained info cannot be read back, the info will be lost. Are you also seeing a state loss if there is no change in MQTT server and no power-down?

Staacks, 2024-06-04:
Well, I think we can stop investigating at "retained messages" :) The restart after which I had to go through the procedure was the one I did to get rid of old entities in HA. For this I deleted all retained messages of the bridge on my MQTT broker.

Arnold-n, 2024-06-04:
The bridge now can delete its own (or all) retained and then rebuild its MQTT messages  (`D12` for own or `D14`for all, in HA: `MQTT_Delete_Own_Rebuild` or `MQTT_Delete_All_Rebuild`). But it also invalidates RTC memory in the process so it still loses state; upon reconsideration that is not really needed so let me remove that from a next release, then it should keep state over `D12`/`D14`.

Arnold-n, 2024-06-05:
In v0.9.53 the tank-installation and cooling-capacity are both assumed present until field setting detection is done. After this upgrade (*) a new round of field setting invitations may be necessary to update the bridge.
(*) and in some of the future upgrades if the version nr of the internal data structure increases.

miguelangel-nubla, 2024-06-05:
Tested and working as expected, thanks!

closed 2024-06-05 by miguelangel-nubla