Arnold-n/P1P2MQTT #103: hc_bridge_v_interface not handled correctly

nash1975, 2024-06-04:
This sensor is flooding my logbook in Home Assistant when the value changes (e.g from 14.7V to 14.8V)
It is somehow not handled as a numeric value but more as a string?
At least when I click on this entity no graph is showing but only the history (Verlauf)

Arnold-n, 2024-06-04:
Thanks for reporting. Which P1P2MQTT-bridge version are you running? V0.9.49 introduced a hysteresis check to reduce the flooding. Only changes of at least 0.2V should be reported now.
I'll make the entity a true "voltage" one in the next release so a graph can be made.

Arnold-n, 2024-06-05:
In v0.9.53 it is formatted as a voltage. Only the icon is still missing, I didn't yet look for a nice mdi image.