Arnold-n/P1P2MQTT #104: sensor.hc_mode_date_time_daikin is flooding my logbook with every second update

nash1975, 2024-06-04:
This sensor is reflecting the date and time of the heatpump.
It is changing every second and by this flooding the history

Arnold-n, 2024-06-04:
There are so many history changes, so I wonder which ones we should limit, and which not? I like to have the correct time visible on a dashboard and do not use history. For the longer term I'd like to make everything configurable using config-files on the ESP, for now I wonder if this justifies a lower update frequency, or alternatively making an option to have the entity disappear?

nash1975, 2024-06-05:
For me ignoring the seconds of the date time and updating on minute change would be the sweet spot.
For my setup I deactivated the entity completely as I have no real need of it.

Arnold-n, 2024-06-05:
In v0.9.53 a parameter 48 is added to remove reporting of seconds in the time format. Commands required to use this are `P48 1` followed by `D5`.

WTH70, 2024-10-03:
You could add this to your configuration.yaml file:

logbook:
  exclude:
    entities:
      - sensor.hc_mode_date_time_daikin

nash1975, 2024-10-05:
Thanks, I will try.
I was not aware of this exclude option for the logbook.
I can use also for another entity.