Arnold-n/P1P2MQTT #105: DHW climate entity missing current DWH temperature.

miguelangel-nubla, 2024-06-04:
Entity `climate.hc_dhw_dhw_setpoint` should include the current temp from sensor `sensor.hc_dhw_temperature_r5t_dhw_tank`.

Arnold-n, 2024-06-04:
Indeed, this is very much related to [tank/cooling detection](https://github.com/Arnold-n/P1P2MQTT/issues/102). The current temp is only available for systems with a tank, but the tank is not recognized until the related field setting is seen. Next release will have as a default setting that a tank is available unless a field setting indicates there is no tank.

Arnold-n, 2024-06-05:
In v0.9.53 tank is assumed to be present, which should show R5T, until a field setting is received. On my tank-less system R5T will show -40C.

closed 2024-06-05 by miguelangel-nubla