Arnold-n/P1P2MQTT #109: Some action not taken into account in HA

Tmbao, 2024-06-30:
Hi Arnold, I've got an issue that if multiple commands are sent to the bridge at almost the same time (they're all triggered via a HA automation), one of the command isn't effective. In the example below, I've got 2nd, 3rd and 5th which all update the controller state. Without the 5 second delay (4th action), the 5th seems to be ignored.

[image]

I think in one of the recent changes, some command buffering was added. Wondering if this is due to some race condition?

Arnold-n, 2024-06-30:
Hi Bao, my mistake, the release notes for v0.9.53 indeed included a message that buffering was added, but that change was moved to a later release. It is currently under test. I  will send you the test version by e-mail.

nicx, 2024-10-11:
@Arnold-n could you send me the test version as well? I have the same problem... as a workaround I configured some Wait-Timers in the automation. But this is not reliable.

Arnold-n, 2024-10-12:
Bao received the F-series test version. I am working on the E-series version and will send it to you later.