Arnold-n/P1P2MQTT #110: Sporadic unavailability of all entities in HA

nash1975, 2024-06-30:
I have the problem that sporadically sometimes the sensor is not available at all in HA.
All entities go then to "not available" status.
After a few seconds, minutes or somethimes even hours the entities come back.

For me it seems that this is nothing related directly to the P1P2 HW.
In MQTT explorer I see the relevant data with no issues

As an example 2 screenshots from yesterday evening with MQTT Explorer and HA sensor reading from LWT.

[image]
[image]

Arnold-n, 2024-06-30:
It sounds like HA receives a will message `offline` on `P1P2/L/P1P2MQTT/bridge0`, making all entities unavailable, which a.o. is generated by the MQTT server when there is a longer MQTT disconnect (perhaps due to poor WiFi?) and when HA does not receive an `online` message when MQTT reconnects (although that message should be transmitted with QoS=1). 

Can you check the value of `P1P2/L/P1P2MQTT/bridge0` when this happens, and log via telnet if there are any errors reported (assuming WiFi connection stays in place when MQTT disconnects)? 

As for WiFi signal strength, `P1P2/P/P1P2MQTT/bridge0/A/9/WiFi_RSSI` shows its strength.

nash1975, 2024-06-30:
The `P1P2/L/P1P2MQTT/bridge0` data yesterday was very instable: (See table below)
RSSI I think is not sooo bad.
Telnet showed the complete day today no errors.

Welcome 192.168.178.67 to P1P2MQTT bridge v0.9.53 compiled Jun  5 2024 22:11:01                                                                                                                                                                               for Daikin E-Series
(Use ^] + q  to disconnect.)
No telnet lock - use 'Pxx _lockword' to add magic lock word
[ESP] 2024-06-30 09:09:46 < Telnet 192.168.178.67 connected

[MON] 2024-06-30 09:10:24 Initiating parameter write for packet-type 0x35 parameter nr 0x40 to value 0x1
[MON] 2024-06-30 09:10:24 Initiating parameter write for packet-type 0x35 parameter nr 0x3E to value 0x1
[MON] 2024-06-30 09:10:27 Executing E command for packet 0x35 setting param 0x40 to 0x1
[MON] 2024-06-30 09:10:27 Executing E command for packet 0x35 setting param 0x3E to 0x1
[MON] 2024-06-30 09:10:32 Initiating parameter write for packet-type 0x35 parameter nr 0x40 to value 0x0
[MON] 2024-06-30 09:10:32 Initiating parameter write for packet-type 0x35 parameter nr 0x3E to value 0x0
[MON] 2024-06-30 09:10:35 Executing E command for packet 0x35 setting param 0x40 to 0x0
[MON] 2024-06-30 09:10:35 Executing E command for packet 0x35 setting param 0x3E to 0x0
[MON] 2024-06-30 09:30:00 Initiating parameter write for packet-type 0x36 parameter nr 0x3 to value 0x226
[MON] 2024-06-30 09:30:00 write 0x36 nr 0
[MON] 2024-06-30 09:30:00 Executing E command for packet 0x36 setting param 0x3 to 0x226
[MON] 2024-06-30 12:41:08 Initiating parameter write for packet-type 0x35 parameter nr 0x40 to value 0x1
[MON] 2024-06-30 12:41:08 Initiating parameter write for packet-type 0x35 parameter nr 0x3E to value 0x1
[MON] 2024-06-30 12:41:09 Executing E command for packet 0x35 setting param 0x40 to 0x1
[MON] 2024-06-30 12:41:09 Executing E command for packet 0x35 setting param 0x3E to 0x1
[MON] 2024-06-30 12:41:21 Initiating parameter write for packet-type 0x36 parameter nr 0x3 to value 0x1F4
[MON] 2024-06-30 12:41:21 write 0x36 nr 0
[MON] 2024-06-30 12:41:21 Executing E command for packet 0x36 setting param 0x3 to 0x1F4
[MON] 2024-06-30 19:23:21 Initiating parameter write for packet-type 0x36 parameter nr 0x3 to value 0x226
[MON] 2024-06-30 19:23:23 write 0x36 nr 0
[MON] 2024-06-30 19:23:23 Executing E command for packet 0x36 setting param 0x3 to 0x226
[MON] 2024-06-30 20:19:45 Initiating parameter write for packet-type 0x35 parameter nr 0x40 to value 0x0
[MON] 2024-06-30 20:19:45 Initiating parameter write for packet-type 0x35 parameter nr 0x3E to value 0x0
[MON] 2024-06-30 20:19:45 Executing E command for packet 0x35 setting param 0x40 to 0x0
[MON] 2024-06-30 20:19:45 Executing E command for packet 0x35 setting param 0x3E to 0x0

[image]

`P1P2/L/P1P2MQTT/bridge0`
Datum	Zeit	Status	Zeitdifferenz
30.06.2024	06:44:00	online	-
30.06.2024	00:17:11	offline	-6.45 hours
30.06.2024	00:16:44	online	-27.29 seconds
30.06.2024	00:16:41	offline	-3.36 seconds
30.06.2024	00:16:26	online	-14.06 seconds
30.06.2024	00:13:47	offline	-2.65 minutes
30.06.2024	00:13:17	online	-30.05 seconds
30.06.2024	00:10:41	offline	-2.6 minutes
30.06.2024	00:10:13	online	-28.25 seconds
30.06.2024	00:07:29	offline	-2.73 minutes
30.06.2024	00:07:00	online	-28.78 seconds
30.06.2024	00:07:00	offline	-0.43 seconds
30.06.2024	00:06:49	online	-11.46 seconds
30.06.2024	00:06:48	offline	-0.55 seconds
30.06.2024	00:06:34	online	-13.92 seconds
30.06.2024	00:03:59	offline	-2.58 minutes
30.06.2024	00:03:05	offline	-54.06 seconds
30.06.2024	00:02:34	online	-31.04 seconds
30.06.2024	00:02:21	offline	-12.93 seconds
30.06.2024	00:02:11	offline	-9.98 seconds
29.06.2024	23:56:07	online	-6.08 minutes
29.06.2024	23:55:57	offline	-9.88 seconds
29.06.2024	23:55:53	offline	-3.51 seconds
29.06.2024	23:55:23	offline	-30 seconds
29.06.2024	23:54:59	offline	-23.96 seconds
29.06.2024	23:54:35	offline	-23.88 seconds
29.06.2024	23:54:11	offline	-24.11 seconds
29.06.2024	23:53:47	offline	-24.01 seconds
29.06.2024	23:53:23	offline	-24.03 seconds
29.06.2024	23:52:53	offline	-30.01 seconds
29.06.2024	23:52:23	offline	-29.99 seconds
[... 44 lines cut ...]
29.06.2024	23:04:39	online	-30.24 seconds
29.06.2024	22:57:45	offline	-6.9 minutes
29.06.2024	22:56:46	offline	-59.94 seconds
29.06.2024	22:55:21	offline	-84.06 seconds
29.06.2024	22:54:47	online	-34.85 seconds
29.06.2024	22:53:01	offline	-105.19 seconds
29.06.2024	22:51:58	offline	-63.86 seconds
29.06.2024	22:50:11	offline	-106.83 seconds
29.06.2024	22:49:51	offline	-19.26 seconds
29.06.2024	22:49:27	offline	-24 seconds
29.06.2024	22:48:45	offline	-42.64 seconds
29.06.2024	22:48:21	offline	-23.39 seconds
29.06.2024	22:47:27	offline	-53.99 seconds
29.06.2024	22:45:16	offline	-2.2 minutes
29.06.2024	22:43:03	online	-2.2 minutes
29.06.2024	22:42:46	offline	-17.3 seconds
29.06.2024	22:42:46	offline	-0.44 seconds
29.06.2024	22:39:55	online	-2.84 minutes
29.06.2024	22:37:28	offline	-2.46 minutes
29.06.2024	22:36:34	offline	-53.97 seconds
29.06.2024	22:36:10	offline	-24.06 seconds
29.06.2024	22:29:46	offline	-6.4 minutes
29.06.2024	22:29:22	online	-23.31 seconds
29.06.2024	22:29:22	offline	-0.52 seconds
29.06.2024	22:29:09	online	-12.7 seconds

Arnold-n, 2024-06-30:
Hi @nash1975 ,

Thanks, well I guess you only see the RSSI values when connected, not when disconnected, so some RSSI values may be worse. In my experience around -80  is not good enough, and may trigger disconnects. I cannot explain why the "online" message is not always received properly, but I have experienced that the ESP MQTT client library struggles under higher load and not-so-good WiFi conditions, so I would recommend to see if improving your WiFi signal solves this issue.

If that does not work, I could repeat the "online" message regularly but it feels like a work-around instead of a solution. And if the WiFi disconnects are really so long as reported that would be an issue to solve.

mckar, 2024-10-11:
I have the same issue. My P1P2 is directly located next to my Unifi AP and my Unifi app says the ESP is connected since over 20 days (last 24h RSSI): 
[image]

I then manually set the bridge to online again. Issue started about 3 days ago and I did not change anything on my wireless network configuration - only updated to latest 2024.10 HA.
Was now offline some hours also I could see that data is transmitted to MQTT (so no real connection issue):
[image]

Arnold-n, 2024-10-11:
@mckar do you use the MQTT server from HA? If so, could this be related to a change in the built-in MQTT server in the new HA version since you only experienced this recently?

I have a few Unifi AP's as well, and have tried a lot of different WiFi settings, but never got WiFi really stable here, to the point that I am considering to switch to a different brand. Phones and other WiFi devices (HomeWizards, RPi) disconnect regularly even if signal strength is really good. Also, one user reported that the DHCP rapid-commit functionality of the DNS server caused a connection issue for him, which was solved by switching rapid-commit option off.

I wonder if we are seeing a combination of (1) regular MQTT disconnects (for whatever reason) and (2a) "online" upon reconnect not getting through due to ESP memory shortage (should not happen!) or (2b) "online" not getting through due to some library or other issue. If so, one solution, though ugly and not solving the root problem, would be to have the bridge repeat the online message regularly.

adam-the, 2024-11-03:
I am also experiencing this since a recent HA update, except my entities don't seem to become available again. MQTT_Delete_All_Rebuild fixes it temporarily, with all of the entities becoming unavailable again after about a day.

My issue might be unrelated, as I've only started having issues since maybe HA 2024.10. I'll do some troubleshooting and update here with any findings.

I did notice some comments [here](https://www.home-assistant.io/blog/2024/10/02/release-202410/) about the Mosquito Broker add-on failing to start post-update, so I'm leaning heavily towards a HA change causing this.

Arnold-n, 2024-11-03:
It might be useful to monitor and log all P1P2/# traffic to understand what is going on when entities become unavailable - for example with mosquitto_client.

mckar, 2024-11-06:
Last feedback from me: I did not experience the issue again during the last weeks, so not sure what caused it.

@adam-the You need to check with the MQTT Explorer Addon if the MQTT values are still updated and only the status in P1P2/L/P1P2MQTT/bridge0 is offline - I did manually change it to online and values are shown again in Home Assistant.

olliewalsh, 2024-11-09:
I hit this recently after upgrading from 0.9.28 to 0.9.53. Previously was very stable in spite of the very weak signal (-80 to -90db), access point rules setup to never disconnect this mac address.

No issues since I moved the ESP from inside the heat pump metal enclosure to outside it. Signal is now -70-75db. 

Would guess that with recent firmware,  TX power required when wifi signal is very low may be drawing too much power on P1P2?

Arnold-n, 2024-11-09:
Hi @olliewalsh thanks for reporting this. It is not a power issue (I use an external power supply and also experienced it once). It appears to be that - as @mckard indicates above - the online status message is sometimes not written when it should be written (after a restart, after a MQTT dis/reconnect). This happens when there is not enough memory available for MQTT transmission. This should be resolved in P1P2MQTT-bridge v0.9.55 which wast just released.

olliewalsh, 2024-11-09:
v0.9.55 looks good so far - reconnects reliably when I kick it from the wifi accesspoint, unavaillable for ~18s in HA.

Arnold-n, 2024-11-10:
Thanks, good to hear!

adam-the, 2024-11-25:
I've also not had any issues since upgrading - think this can be closed! Thank you.

closed 2024-11-25 by Arnold-n