Arnold-n/P1P2MQTT #111: Entity climate.hc_LWT Abs_Cooling and climate.hc_LWT Absnot available

MasterRubke, 2024-07-12:
The entity for setting the leaving water temperature is not available.
The only climate entity's available are for deviation.

[image]

Arnold-n, 2024-07-15:
This is  intentional, at least in the current version (subject to discussion on reconfiguration of climate entities). You cannot set LWT yourself if your system operates in weather-dependent mode, only the deviation of LWT as calculated by the WD curve. The Deviation_* climate entities will be reconfigured to LWT_* climate entities if you switch your system from WD mode to ABS mode (which can be done from HA).

Staacks, 2024-07-15:
Not sure if this is the case for @MasterRubke, but I noticed an edge case that might add to the confusion:
On my heatpump one does not set WD and ABS mode globally, but WD/ABS is set independently for heating and cooling.

In my case it was set to WD for heating and ABS for cooling by default (or the installer). The bridge showed the WD setting of heating and disabled the ABS controls for cooling with the result that the heating controls were fine, but I could only control relative temperature of cooling which had no effect as the heatpump was using the absolute setting.

Arnold-n, 2024-07-16:
Thanks, that might explain it. In that case we would need to search for the bits and bytes that report and control WD/ABS mode in cooling mode; I will add some monitoring code to P1P2MQTT-bridge later this week, perhaps this evening, to support this search.

MasterRubke, 2024-07-16:
Putting the system in ABS-mode solved the problem. I can now set the LWT temperature.

Staacks comment is partially true for me. I can not set the heating and cooling modes separate. 
Instead my system has 3 operating modes:
1: ABS
2: WD for heating / ABS for cooling
3: WD
So for me it would be the same entity, only adding a third option "WD/ABS"

Is there a way where i can check the value of the paramter? I think over TELNET but i don't know the command yet.
I am new to this matter. :)

Staacks, 2024-07-16:
Oh, sorry, you are right, it is also not entirely independent for me:
[image]