Arnold-n/P1P2MQTT #114: The "F1" not stored to EEPROM

DiabloVmax1200, 2024-07-15:
As written here: https://github.com/Arnold-n/P1P2MQTT/issues/83#issuecomment-2118903166 the F1 is not stored to EEPROM. Could you please implement this. Would make life easier - each change at the Daikin Unit leads to a restart and resulting in an error because of the "missing" Roomtemp.

Arnold-n, 2024-07-15:
Can you test P1P2Monitor v0.9.54? It should store the setting in EEPROM and should fix the 25.6C bug.

DiabloVmax1200, 2024-07-15:
Thanks EEPROM Storage of F1 State is also fixed with Firmware_images/P1P2Monitor-v0.9.54-Eseries-P1P2MQTT-bridge.ino.hex

closed 2024-07-15 by DiabloVmax1200

Arnold-n, 2024-07-15:
Great, many thanks for your quick testing and your kind sponsoring!

DiabloVmax1200, 2024-07-15:
I have to thank. Great work!