Arnold-n/P1P2MQTT #117: Telnet session not possible after several days, non-heating mode

buuhsmead, 2024-07-21:
I normally use the telnet session to send commands. After two days in mode 'E35002F00' and not using telnet I am not able to reconnect with a telnet session.
The only option I have is to write a mode change from LWT to RT-ext/RT onto 'P1P2/W/P1P2MQTT/bridge'.
And because of that, the MMI is resetting the system and dropping the power on P1P2 lines.
The system gets a reboot. 
After that, it works again.

Can I do something else to get a telnet session.
I tried to write 'K' to 'P1P2/W/P1P2MQTT/bridge', but that did not work.

Arnold-n, 2024-07-21:
The bridge allows only 1 telnet session at any point in time. A 2nd connection will fail if the first is still active, but will drop the 1st too, and the next telnet attempt should work again. Could this behaviour explain your observations?

As you have only 1 bridge, it is easier to write `K` to P1P2/W (easier). The correct full topic is P1P2/W/P1P2MQTT/bridge0, with a 0 at the end. But 'K' resets the ATmega, which does not help to restore telnet. If you write `D0` instead of `K` you will reset the ESP which should re-enable telnet if telnet would not work.

buuhsmead, 2024-07-23:
Update:
Again 24-36 hours not connected to telnet, and not sure if cleanly stop the telnet session. A telnet session was not possible. But I will try to test a little more and cleanly stop the telnet session. And get back with an update.

What did help was writing 'D0' to 'P1P2/W/P1P2MQTT/bridge0'. I was able to get a telnet session.
So that part is solved, thanks Arnold.

Arnold-n, 2024-07-23:
I currently use v2.1 of the ESP Telnet library, I'll try to upgrade to 2.2.1 to see if that solves your issue.

buuhsmead, 2024-07-24:
I made sure that I correctly closed the telnet session. Currently, no hick-ups getting connected.
Think it is just PEBKAC.

Closing this for now. If needed, will re-open it.
Thanks for your support, Arnold.

closed 2024-07-24 by buuhsmead