Arnold-n/P1P2MQTT #122: Heating shows Enabled when system is off, and dual states

matthijsberg, 2024-08-01:
Hi, 

I'm looking at  HC_Mode Climate_Heating (or Cooling or Auto) in a E series HP. When I enable this (as in enabled, not actually active / running / heating) state, like heating, in the app, this gets reflected in the system, however there are two situations I'd like to run by you for discussion;

1. There are 3 states that are unique (heat / cool / auto). When I enable auto, the last other active state is also still active in HASS (example, Heat was active, when I select Auto, Heat remains active, but Cool does not become active). I would expect ONLY auto to be active, not heat and / or cool. 
2. When a state is active (let's say heat) and I disable the complete heatpump (the "power" button in my app, basically disabling all heating / cooling) the active state remains the last known state. I would expect it to disable all states as the complete system is off. (DHW is seperate from this by the way). 

Would it make sense to build this logic in? 

thnx, Matthijs

Arnold-n, 2024-08-01:
Hi Matthijs,

Climate_Heating/Cooling is the information directly provided by the Daikin indoor unit. It reports in which mode the system is, even if off. If you select heat/cool  mode, this should be reflected, but in auto mode, the system will select heating/cooling based on outside temperature, taking a hysteresis into account. To me it makes sense that these can be active even if the system is off, similar to the fact that the system can be on even if the compressor if off (during start-up, or due to LWT overshoot or due to a too-high (in heating) or too-low (in cooling) return water temperature).

If you want a radio-button type behaviour, it is provided in the `Heating_Cooling_Auto` select entity itself.

matthijsberg, 2024-08-02:
Thanks, makes sense. 

Just a FYI, when selecting auto, coming from heat / cool, that last state remains on too in my experience so far. This does not seem to be an informed decision by the system.

closed 2024-08-02 by matthijsberg