Arnold-n/P1P2MQTT #123: Hitachi H-link2 - support for writing to the P1P2 bus

user890104, 2024-08-10:
We have a Hitachi H-link2 system installed at our office. There is a P1P2MQTT bridge made by Arnold that we installed and we can see the raw packets going back and forth. But we are unable to control the units, since writing to the bus is not implemented.

We also have a spare remote control, but unsure if it can be of any use for debugging/reversing the protocol needed for write support on the bus.

This is our reverse-engineering effort so far, based on raw packet sniffing, and chaning the system to run in various modes:
https://docs.google.com/spreadsheets/d/13hszgDAdVjzlA0g9ORPiE48NfMgoU1kxcpj9PWbSJ1w/edit

Please let me know if anything else is needed so that writing to the bus e.g. for replaying command from the remote, can be achieved.

Arnold-n, 2024-08-11:
Hi Vencislav,
Many thanks for your support in reverse engineering this - I am very busy in August but will pick up Hitachi support in September again.
For other Hitachi users: please upgrade to at least v0.9.51 which contains some decoding based on Vencislav's analysis.
Kind regards,
Arnold

user890104, 2024-10-02:
Hi, just a kind reminder on this. As I said previously, we are open to help understanding the protocol, including hooking up a logic analyzer to the bus and/or sniffing the raw data from the original remote. We don't have access to the central management software, just to our office's units and wall remote control panel, and this is why we need remote access to configure the HVAC system.

linhpd88, 2025-05-16:
@user890104 what is the progress of this, can you share something?

daniyo27, 2025-07-31:
Hi @Arnold-n !
Is there any progress on writing to the P1P2 bus? I have a Hitachi system and would really love to get some sort of WiFi control on it. 

I bought a chinese thermostat Acmelec, and it seemed to do basic functions such as heating, cooling, dry, and fan, but the fan control did not support auto mode nor the fourth fan setting which is the highest.

If I can be of assistance somehow, I would love to help out.

Arnold-n, 2025-08-03:
We know that sending commands to the indoor unit works but, not fully reliably, but we do not know how bus arbitration works to avoid bus collission. It would be really nice to understand how writes should be timed on the bus.

remynicol, 2025-08-22:
No, I didn’t need an oscilloscope — the low-level part of reading the Hitachi bus was already very well implemented.  
In particular, the handling of double bits (UC errors) and parity bit errors (PE).  
The only missing piece was the emulation of a remote control in order to “drive” the physical remote (the messages with header `0x41`).  

Displaying the PE errors (by changing the *E* mask) allowed me to see that my Hitachi heat pump correctly sends all parity bits, whereas my remote controls systematically invert the parity bit of the first byte.  

And if we want to emulate a remote control, we need to reproduce this behavior, otherwise the heat pump ignores the message.  

That’s the lowest-level work I had to do — but using the P1P2MQTT hardware, so.

linhpd88, 2025-08-25:
> I believe this message is how a new remote control obtains its addressing on the bus. It sends this special packet when it connects for the first time. The main remote then replies with a packet where the data field contains the remote ID that the new RC should use. In other words, the main remote assigns the ID to the newcomer.

You beleave is not a proof, that why I ask about data capture from the bus, and to act like a sub controler, your code is not finish. To make it act as Sub, you must remove 1 RC and simulate like the other, you have PC-ARFP1E (2020) + PC-ARFG2-E (2024), now you need to keep 1, example PC-ARFP1E (2020) set as Main RC, then code the firmware to make the p1p2mqtt act like the PC-ARFG2-E (as Sub RC).

linhpd88, 2025-08-26:
> Exactly what I did. I have only one remote and the p1p2. Again tonight, I was able to turn on the air conditioning remotely. Let me know if you have any difficulties testing it.

I have the York one, and your code not working with it, So I try to understand the code to adapt it with try/fail. But there are some place I do not understand it, that why I ask. You can point me some importance place of your current PR? I do not care abour decode data for HA, just the part to make p1p2mqtt as secondary remote.

linhpd88, 2025-08-29:
> I see what you mean 🙂 In my setup, it’s at least necessary to handle the `21001C` frames, since they are used to complete the `41` messages that **p1p2mqtt** will generate.
> 
> Could you share a Telnet capture of a few minutes with the following parameters ?
> 
> J12 E3F
> 
> If possible, please also include one or two write attempts during the capture.
> 
> From what I’ve learned with the Hitachi model 1 and 2 implementations, it looks like there are almost as many frame variations as there are units 🤷. Having a look at the frames from your PAC would really help us target directly the ones that are causing trouble.

It only have:
21 00 1A, 21 00 30, 21 00 1B and 21 1f 0a, there is no 89 00 2D data or it failed to decode, Can you tell me your email?