Arnold-n/P1P2MQTT #127: Link Between outside and inside units

x3oo, 2024-09-12:
Hi does anybody know if the link between the outside and inside units use the same protocoll? Usually many ACs only use one wire for communication between those.

Arnold-n, 2024-09-15:
Some AC units use 2-wire F1/F2 which is electrically the same as P1/P2, but logically different. And F1/F2 may not provide bus power.

Other units use a one-wire protocol, this may be unsafe to touch, and which is definitely different from P1/P2.

x3oo, 2024-09-16:
For context:
Many ACs are well capable of heating up your water based heating system, if only you could attach a heat exchanger as an indoorunit. Since many people already do have AC Units it would be quite easy to connect them to the heating system if only you could communicate with the Outside unit.

Another application would be to build some specialised systems for example to heat your RV and warm water.

Also prices for Waterunits seem to be quite high compared to Air/Air Units.

x3oo, 2024-09-16:
> Some AC units use 2-wire F1/F2 which is electrically the same as P1/P2, but logically different. And F1/F2 may not provide bus power.
> 
> Other units use a one-wire protocol, this may be unsafe to touch, and which is definitely different from P1/P2.

Daikin Emura uses only one wire to my knowledge.

x3oo, 2024-09-17:
> Some AC units use 2-wire F1/F2 which is electrically the same as P1/P2, but logically different. And F1/F2 may not provide bus power.

Logically only means that there are other messages or that it's another protocol?

Arnold-n, 2024-09-17:
It's also packet-based, but with different packet lenghts and (unknown) content.

x3oo, 2024-09-17:
> It's also packet-based, but with different packet lenghts and (unknown) content.

how hard would it be for a motivated human being to decode?

Arnold-n, 2024-09-18:
Certainly doable!