Arnold-n/P1P2MQTT #128: Problem set heat temperature

mochitoju, 2024-09-14:
Hi there, 

i've the "modul" now since 2 month for my Daikin Altherma 3 (new model) and now its the first time to start testing heating. 
i can not set any heating temperature (Room Room_Heating) above 20,4 degrees.
-> "only allowed between 12 and 20,4 degrees"

in cooling i can set temperature till 27 degrees.

anybody who can help?
in Daikin Onecta HomeAssistant it would be possible to set the correct temperature till 32 degrees.

gtx, juergen

mochitoju, 2024-09-14:
sensor.bridge1_fieldsettings_3_06_a3212_temp_range_room_heating_max
= 20.4 

so, its obtained from Daikin, but i cannot find where i can change it

Arnold-n, 2024-09-14:
Hi Juergen,

You cannot change this value - by coincidence I learned yesterday that some field settings, in particular the stepsize for certain field settings, are not decoded properly on some systems, and I think this is the cause of your problem too.

Can you please share the MQTT reported values of the following 4 (or 8) settings?
```
P1P2/P/P1P2MQTT/bridge0/F/2/3_06_A3212_Temp_Range_Room_Heating_Max
P1P2/P/P1P2MQTT/bridge0/F/2/3_07_A3211_Temp_Range_Room_Heating_Min
P1P2/P/P1P2MQTT/bridge0/F/2/3_08_A3214_Temp_Range_Room_Cooling_Max
P1P2/P/P1P2MQTT/bridge0/F/2/3_09_A3213_Temp_Range_Room_Cooling_Min
P1P2/P/P1P2MQTT/bridge0/F/4/3_06_A3212_Temp_Range_Room_Heating_Max
P1P2/P/P1P2MQTT/bridge0/F/4/3_07_A3211_Temp_Range_Room_Heating_Min
P1P2/P/P1P2MQTT/bridge0/F/4/3_08_A3214_Temp_Range_Room_Cooling_Max
P1P2/P/P1P2MQTT/bridge0/F/4/3_09_A3213_Temp_Range_Room_Cooling_Min
```

On my system these report:
```
P1P2/P/P1P2MQTT/bridge0/F/2/3_06_A3212_Temp_Range_Room_Heating_Max {"val":30.0, "min":18, "max":30, "stepsize":0.5, "bits":"00", "bitsstep":"0x01"}
P1P2/P/P1P2MQTT/bridge0/F/2/3_07_A3211_Temp_Range_Room_Heating_Min {"val":12.5, "min":12, "max":18, "stepsize":0.5, "bits":"00", "bitsstep":"0x01"}
P1P2/P/P1P2MQTT/bridge0/F/2/3_08_A3214_Temp_Range_Room_Cooling_Max {"val":35.0, "min":25, "max":35, "stepsize":0.5, "bits":"00", "bitsstep":"0x01"}
P1P2/P/P1P2MQTT/bridge0/F/2/3_09_A3213_Temp_Range_Room_Cooling_Min {"val":15.0, "min":15, "max":25, "stepsize":0.5, "bits":"00", "bitsstep":"0x01"}
```
with a stepsize of 0.5C (which is configurable on my system as 0.5C or 1.0C). I think your system has a stepsize of 1.0C (fixed) whereas it seems to be wrongly decoded as 0.2C. I would be interested to see the value of your "bits" and "bitsstep" in these messages to see if they are needed in the field setting decoding.

mochitoju, 2024-09-14:
3_06_A3212_Temp_Range_Room_Heating_Max = {"val":20.4, "min":18, "max":20, "stepsize":0.2, "bits":"00", "bitsstep":"0x40"}
3_07_A3211_Temp_Range_Room_Heating_Min = {"val":12.0, "min":12, "max":13, "stepsize":0.2, "bits":"00", "bitsstep":"0x40"}
3_08_A3214_Temp_Range_Room_Cooling_Max = {"val":27.0, "min":25, "max":27, "stepsize":0.2, "bits":"00", "bitsstep":"0x40"}
3_09_A3213_Temp_Range_Room_Cooling_Min = {"val":15.0, "min":15, "max":17, "stepsize":0.2, "bits":"00", "bitsstep":"0x40"}

mochitoju, 2024-09-14:
mhm, i think i could change the value to more than 20 in the beginning (also a friend of mine have the same problem) - maybe something changed according a firmware update of Daikin?

Arnold-n, 2024-09-14:
Quite possibly - it seems Daikin now uses a mantissa which can be 1.0 which I haven't seen in the past. I am updating the firmware and hope that will solve it.

mochitoju, 2024-09-14:
little update.
- rebuild mqtt
- changed field_settings > RT > LT and back to RT 
- temperature range to 30 degrees possible!

Arnold-n, 2024-09-14:
Seems weird, but I think this happens because the rebuild-mqtt action (with current 0.9.53) will work only for a few minutes (it uses 30 as a best-guess value, while waiting for the real max) before it then changes back to 20.4.

Just mailed you a new firmware 0.9.54, which should solve it. Daikin seems to use 10E-1 instead of 1E0 for certain step sizes.

Arnold-n, 2024-09-15:
[P1P2MQTT-bridge-v0.9.54rc1-Daikin-E.ino.bin](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.54rc1-Daikin-E.ino.bin) should decode 10E-1 step size correctly.

mochitoju, 2024-09-15:
> Seems weird, but I think this happens because the rebuild-mqtt action (with current 0.9.53) will work only for a few minutes (it uses 30 as a best-guess value, while waiting for the real max) before [...]

Problem seem to be solved. 

> [P1P2MQTT-bridge-v0.9.54rc1-Daikin-E.ino.bin](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.54rc1-Daikin-E.ino.bin) should decode 10E-1 step size correctly.

is this the same file or a diffferent one/version?

Thank you much for your quick solution!

Arnold-n, 2024-09-15:
It's the same, except for the version itself.

Excentyl, 2025-03-22:
I am having an issue (on firmware 0.9.58rc13) which I think might be the same as this one? I am not able to set my temperature because my min_temp is wrong, here is the attributes of the cliamte entity:
```
hvac_modes:
  - "off"
  - heat
min_temp: 212
max_temp: 26
target_temp_step: 0.5
current_temperature: 19.9
temperature: 17
icon: mdi:coolant-temperature
friendly_name: Central Heating
supported_features: 385
```

I checked the setting of F/F/3_07_A3211_Temp_Range_Room_Heating_Min and it looks quite strange:
```
{
    "val": 212,
    "min": 12,
    "max": 612,
    "stepsize": 100,
    "bits": "00",
    "bitsstep": "0x00"
}
```

On the MMI the setpoint range is correct, but if I change it to something else and then change it back, the climate entity in Home Assistant will correct itself and I can change the temperature setpoint again. However, after a little while, the problem will reoccur - I have so far been unable to determine what causes this (note: sometimes it effects the max temperature instead).

Also, if you'd rather, I can open a new issue for this, just let me know.

Arnold-n, 2025-03-22:
@Excentyl thanks for reporting this. I think it has the same root cause, and I will rewrite the code to be more lenient on stepsize used.

Arnold-n, 2025-03-22:
The cause is likely somewhat similar, but still a bit strange. This version has a work-around which I hope works better: [P1P2MQTT-bridge v0.9.58rc14 for Daikin-E](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.58rc14-Daikin-E.ino.bin)

Excentyl, 2025-03-22:
Thank you @Arnold-n, I have installed the new firmware and after resetting the setpoints in the MMI, the climate entity in Home Assistant is working and F/F/3_07_A3211_Temp_Range_Room_Heating_Min looks correct:
```
{
    "val": 14,
    "min": 12,
    "max": 18,
    "stepsize": 1,
    "bits": "00",
    "bitsstep": "0x00"
}
```

Since I'm not sure what was causing the problem to reoccur before, I'll keep and eye on it and let you know if there are any further issues.

Thank you again for your support.

Arnold-n, 2025-03-22:
You're welcome. You mentioned earlier that the problem would,  "after a little while", reoccur. Do you know what time frame this was, less than 10 minutes, or if longer, perhaps correlated to a Daikin system restart? As I think the problem could be triggered
-during a restart
-during a regular repetition of field settings values (which does not happen on all systems)
-or due to another cause

Excentyl, 2025-03-24:
I haven't had a reoccurance over the weekend - before the update on Friday, it was happening a couple times a day, which gives me confidence. Now that you mention it, it may be linked to system restarts.

I first noticed this after I had power monitoring installed (a few days after the heat pump was installed). I initially didn't think that would cause an issue since it's just a CT clamp on a wire, but the system did get powered down and then restarted to install it. Following that, I was still regularly changing settings on the MMI (set points, weather curve ect) and the MMI did occationally ask to restart the system. I also changed settings in the MMI over the weekend without issue.