Arnold-n/P1P2MQTT #136: UDP or ESPHome

uKL, 2024-10-24:
Hello,
Are you considering adding support for UDP based communication or creating ESP Home based configuration?

Arnold-n, 2024-10-24:
Yes, but several other topics have higher priority, so sorry but this may take a while.

DiabloVmax1200, 2024-10-24:
uKL for what do you plan to use UDP based communication? Because of Loxone integration? If yes Loxberry with MQTT Sever works pretty. I also first thought UDP would be great to have. But the MQTT integration has huge benefit for Loxone. https://wiki.loxberry.de/konfiguration/widget_help/widget_mqtt

uKL, 2024-10-27:
> uKL for what do you plan to use UDP based communication? Because of Loxone integration? If yes Loxberry with MQTT Sever works pretty. I also first thought UDP would be great to have. But the MQTT integration has huge benefit for Loxone. https://wiki.loxberry.de/konfiguration/widget_help/widget_mqtt

Yes, for Loxone. I'd be using MQTT for now but for stability I'd like to make it independent of my Loxberry Instance like I have it for other ESP32 based sensors.

DiabloVmax1200, 2024-10-27:
Hi. Where do you have stability issues? I’ve the same setup but actually have the feeling that it runs stable

uKL, 2024-10-27:
> Hi. Where do you have stability issues? I’ve the same setup but actually have the feeling that it runs stable

In general it is stable. Conceptually, I’m having an idea of making critical items like security, heating, electricity, independent of Loxberry - just in case.

DiabloVmax1200, 2024-10-28:
Hi, this makes sense. Less parts more reliable.
In my configuration I only would lose some comfort and I might would use a bit more energy.

uKL, 2024-11-07:
> Hi, this makes sense. Less parts more reliable. In my configuration I only would lose some comfort and I might would use a bit more energy.

Well, it depends. If the heatpump would be stuck in DHW heating disabled, this means that there's no hot water in the shower ;-) Wife acceptance factor = -1

... and his had happened with the cloud integration for a few times.

DiabloVmax1200, 2024-11-07:
That’s definitely an issue ☺️

Gorgar91, 2024-11-17:
I purchased "older" model just because of lack of UDP functionality, using budulinek setup now with loxone, but if UDP will come into place , I am ready to do purchase again .  Thanks Arnold for all this !