Arnold-n/P1P2MQTT #145: Altherma 3 - ETBX12EF9W - No installer mode, no silent mode

Rewertius, 2024-11-27:
Hello,
first of all: thank you very much for you important work to emancipate from the cloud and the closed source software.
After a little struggle with the setup, i am able to control the main features ob my ETBX12EF9W.

Now i would like to monitor the behavior of the heatpump and the bridge. Therefore i try to find the mqtt topics as described in your manual. But in my mqtt explorer i can only see the topics as you see in my attached screenshot
[image]

I would like to the these informations because if i try to change the silent mode or the user mode by using the entitys in homeassistant , it does'nt have any impact. Also if use the L92 commant via telnet.

thanks in advance for any hints.
Philip
PS: if this post is more a kind of a discussion than an issue, fell free to move.

Rewertius, 2024-11-27:
Now i found an entry as using the L92 command: 
[image]
but nothing after that.

Arnold-n, 2024-11-27:
This is likely the same issue as [issue #98](https://github.com/Arnold-n/P1P2MQTT/issues/98) - it seems that the bridge attempting to change quiet mode or trying to claim installer mode does not work on all systems - not sure why yet. Perhaps it is not possible, perhaps the mechanism is different.