Arnold-n/P1P2MQTT #148: Is climate.set_preset_mode available?

fauust, 2024-12-03:
Hi!

First of all, this is an awesome project! I have just received the bridge (well packed) and thanks to the documentation, it was pretty straightforward to setup and to start using it (I had prepared the MQTT part). Thanks a lot @Arnold-n and anybody that has contributed to this project!

I have a very simple question, is the preset mode available `climate.set_preset_mode`?

Here are HA logs:
```console
homeassistant  | 2024-12-03 18:13:11.271 ERROR (MainThread) [homeassistant.components.automation.tdegc_lavoro_09h00_17degc] Error while executing automation automation.tdegc_lavoro_09h00_17degc: Entity climate.hc_room_room_heating does not support this service.
```

The reason I am trying to use it is to synchronize what the Daikin LCD screen shows with what HA actually does (temperature targets are in sync though which is already awesome).

I am very sorry if this was already discussed (I could not find any issue), or if it's explained in the somewhere documentation.

fauust, 2024-12-04:
Thanks!

Yes, I am referring to the eco/comfort state of the controller. The use case would be someone (a guest) without HA access that could still easily understand what's going on on the Daikin LCD and make changes between confort and eco, which he still can but need to check the desired temperature to understand the HA program.

But I can totally live without it!

Arnold-n, 2024-12-04:
Thanks for your explanation. I will add a schedule/eco/comfort selection menu in the next version. However I think that it only shows the manual preset (=override), but not what the schedule itself prescribes, this can only be derived from the actual  room temperature setpoint.

fauust, 2024-12-05:
Thanks! BTW is there a way to be alerted of new firmware release, I see that the repo has only 3 tags so you probably don't release through GitHub?

Arnold-n, 2024-12-06:
[v0.9.58rc3](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.58rc3-Daikin-E.ino.bin) includes preset mode. Firmwares are frequently released via github, but usually not tagged as most versions are only incremental. I think you can "watch" a repo to be informed of any commits (some of which are firmware).

closed 2024-12-10 by fauust

Arnold-n, 2024-12-10:
Thanks for providing this script - but why 2 actions, isn't the 2nd action enough, the wall controller should change the temperature setting to the value configured for eco when the preset mode is set to eco.

fauust, 2024-12-11:
You are right, it is but I don't want to rely on the wall setup so that I can change Eco/Comfort temp from HA. But next step is to make those changes in sync between HA and Daikin so that only preset_mode is needed :)

fauust, 2024-12-13:
Awesome!

fauust, 2024-12-16:
I have just upgraded the firmware, it's working like a charm! Best couple of euros spend in 2024 ;)