Arnold-n/P1P2MQTT #152: Duplicate HA entries created

brainrecursion, 2024-12-06:
After setup I get a few duplicate entries created:
- climate.hc_room_room_cooling
- climate.hc_room_room_cooling_2
- climate.hc_room_room_heating
- climate.hc_room_room_heating_2

Also is it possible to add a config option to not create any of the cooling entities if the P1P2 device does not support cooling?
Thanks

Arnold-n, 2024-12-06:
Did you change any config options (like MQTT prefix)? If not, can you share the related config items (before deleting them)? You can delete no longer used entities with `MQTT_Delete_Own_Rebuild`. or even `MQTT_Delete_All_Rebuild`.

Deciding whether to make cooling entities is tricky, as the cooling capability - a field setting - is only reported by the Daikin system once, shortly after system restart, and in many but not all systems also every regularly thereafter, every 5-10 minutes. In an earlier version cooling entities were deleted once it was detected that a system is heating-only but that wasn't working reliably with the way HA shows various entities so they are currently not deleted. For now the next-best solution is to delete the entities from your dashboard.

brainrecursion, 2024-12-06:
> Did you change any bridge config options (like MQTT prefix)? If not, can you share the related config items (before deleting them)? You can delete no longer used entities with `MQTT_Delete_Own_Rebuild` or even `MQTT_Delete_All_Rebuild`.

I haven't changed any bridge config options. The duplicate entries were after running `MQTT_Delete_Own_Rebuild`. I just ran it again and got a different set of duplicate entries, so something strange going on.

Arnold-n, 2024-12-07:
Not only changing MQTT prefix (configuration P12), but also changing the device name (P13, default P1P2MQTT), changing the bridge name (P14, default bridge0), whether to use device name in topic (P24), whether to use bridge name in topic (P28), and perhaps also whether to use bridge name in entity, may give duplicate entries. Would that explain your situation? In that case `MQTT_Delete_Own_Rebuild` will create a new entry without deleting the original one (as it no longer recognizes the original entry as its own), and `MQTT_Delete_All_Rebuild` is needed.

rjpearce, 2024-12-07:
I have this issue too with the following duplicate entries:
- climate.hc_room_room_cooling
- climate.hc_room_room_cooling_2
- climate.hc_room_room_cooling_3
- climate.hc_room_room_heating
- climate.hc_room_room_heating_2
- climate.hc_room_room_heating_3

I hadn't changed P12, P13, P14, P24 or P28. 
I turn L1 on, off and then back on, I turned on C2 and change some of the Field Settings.

I can confirm that MQTT_Delete_All_Rebuild (command D14) does remove them all from HC_Room but I now have the following back:

- climate.hc_room_room_cooling
- climate.hc_room_room_cooling_2
- climate.hc_room_room_heating

brainrecursion, 2024-12-07:
I haven't changed P12, P13, P4, P24 or P28. 
I have tried `MQTT_Delete_Own_Rebuild` a few times and I can see it removes all the entities but I end up with a few random duplicates each time. Is it something simple like the MQTT topic config messages are not being sent with a QoS of 2 so are received duplicated? Equally it could be a HomeAssistant bug.

Arnold-n, 2024-12-07:
Perhaps its HA. I sometimes had to delete the entire MQTT configuration and add it again in HA to resolve things. Though duplicated messages should have the same uniq id and should not give duplicate entries. If you have mosquitto_sub, you can read back all retained homeassistant config messages to check if there are duplicated config messages, as in
`mosquitto_sub -h 192.168.4.12  -p 1883  -u P1P2 -P P1P2 -v -t homeassistant/# | grep Room_Cooling`. This will also show any messages get deleted and reconfigured during a rebuild.

If your WiFi signal is weak, that could cause loss of some messages too due to limited buffer sizes.

brainrecursion, 2024-12-07:
I tried the `mosquitto_sub ...` command and it did return three entries which matches the amount of entities I am seeing. Interesting they all have different unique IDs.

```
homeassistant/climate/bridge0/P1P2MQTT_bridge0_U4_Room_Cooling_4/config {"temp_stat_t":"P1P2/P/P1P2MQTT/bridge0/S/0/Room_Cooling_Setpoint","min_temp":15.0,"max_temp":35.0,"temp_step":0.5,"curr_temp_t":"P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_Room","mode_stat_t":"P1P2/P/P1P2MQTT/bridge0/S/0/Altherma_On","mode_stat_tpl":"{% set modes={'0':'off','1':'cool'} %}{{ modes[value] if value in modes.keys() else modes['0']}}","modes":["off","cool"],"mode_cmd_t":"P1P2/W/P1P2MQTT/bridge0","mode_cmd_tpl":"{% set modes={'off':0,'cool':1} %}{{'E35002F%02X 350031%02X 35002D%02X'|format((modes[value]|int) if value in modes.keys() else 0, (modes[value]|int) if value in modes.keys() else 0, (modes[value]|int) if value in modes.keys() else 0)}}","temp_cmd_t":"P1P2/W/P1P2MQTT/bridge0","temp_cmd_tpl":"{{'E360001%04X'|format((value*10)|int)}}","name":Room_Cooling","uniq_id":"P1P2MQTT_bridge0_U4_Room_Cooling_4","avty":[{"topic":"P1P2/L/P1P2MQTT/bridge0","pl_avail":"online","pl_not_avail":"offline"},{"topic":"P1P2/P/P1P2MQTT/bridge0/A/8/Control_Function","pl_avail":1,"pl_not_avail":0},{"topic":"P1P2/P/P1P2MQTT/bridge0/C/9/RT","pl_avail":1,"pl_not_avail":0},{"topic":"P1P2/P/P1P2MQTT/bridge0/S/9/Heating_Only","pl_avail":0,"pl_not_avail":1}],"avty_mode":"all","dev":{"name":"HC_Room","ids":["HC_Room"],"mf":"Home-CliCK","mdl":"P1P2MQTT_bridge","sw":"0.9.58rc2"},"ic":"mdi:coolant-temperature","qos":1}
homeassistant/climate/bridge0/P1P2MQTT_bridge0_U6_Room_Cooling_6/config {"temp_stat_t":"P1P2/P/P1P2MQTT/bridge0/S/0/Room_Cooling_Setpoint","min_temp":15.0,"max_temp":35.0,"temp_step":0.5,"curr_temp_t":"P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_Room","mode_stat_t":"P1P2/P/P1P2MQTT/bridge0/S/0/Altherma_On","mode_stat_tpl":"{% set modes={'0':'off','1':'cool'} %}{{ modes[value] if value in modes.keys() else modes['0']}}","modes":["off","cool"],"mode_cmd_t":"P1P2/W/P1P2MQTT/bridge0","mode_cmd_tpl":"{% set modes={'off':0,'cool':1} %}{{'E35002F%02X 350031%02X 35002D%02X'|format((modes[value]|int) if value in modes.keys() else 0, (modes[value]|int) if value in modes.keys() else 0, (modes[value]|int) if value in modes.keys() else 0)}}","temp_cmd_t":"P1P2/W/P1P2MQTT/bridge0","temp_cmd_tpl":"{{'E360001%04X'|format((value*10)|int)}}","name":Room_Cooling","uniq_id":"P1P2MQTT_bridge0_U6_Room_Cooling_6","avty":[{"topic":"P1P2/L/P1P2MQTT/bridge0","pl_avail":"online","pl_not_avail":"offline"},{"topic":"P1P2/P/P1P2MQTT/bridge0/A/8/Control_Function","pl_avail":1,"pl_not_avail":0},{"topic":"P1P2/P/P1P2MQTT/bridge0/C/9/RT","pl_avail":1,"pl_not_avail":0},{"topic":"P1P2/P/P1P2MQTT/bridge0/S/9/Heating_Only","pl_avail":0,"pl_not_avail":1}],"avty_mode":"all","dev":{"name":"HC_Room","ids":["HC_Room"],"mf":"Home-CliCK","mdl":"P1P2MQTT_bridge","sw":"0.9.58rc2"},"ic":"mdi:coolant-temperature","qos":1}
homeassistant/climate/bridge0/P1P2MQTT_bridge0_U2_Room_Cooling_2/config {"temp_stat_t":"P1P2/P/P1P2MQTT/bridge0/S/0/Room_Cooling_Setpoint","min_temp":15.0,"max_temp":35.0,"temp_step":0.5,"curr_temp_t":"P1P2/P/P1P2MQTT/bridge0/T/1/Temperature_Room","mode_stat_t":"P1P2/P/P1P2MQTT/bridge0/S/0/Altherma_On","mode_stat_tpl":"{% set modes={'0':'off','1':'cool'} %}{{ modes[value] if value in modes.keys() else modes['0']}}","modes":["off","cool"],"mode_cmd_t":"P1P2/W/P1P2MQTT/bridge0","mode_cmd_tpl":"{% set modes={'off':0,'cool':1} %}{{'E35002F%02X 350031%02X 35002D%02X'|format((modes[value]|int) if value in modes.keys() else 0, (modes[value]|int) if value in modes.keys() else 0, (modes[value]|int) if value in modes.keys() else 0)}}","temp_cmd_t":"P1P2/W/P1P2MQTT/bridge0","temp_cmd_tpl":"{{'E360001%04X'|format((value*10)|int)}}","name":Room_Cooling","uniq_id":"P1P2MQTT_bridge0_U2_Room_Cooling_2","avty":[{"topic":"P1P2/L/P1P2MQTT/bridge0","pl_avail":"online","pl_not_avail":"offline"},{"topic":"P1P2/P/P1P2MQTT/bridge0/A/8/Control_Function","pl_avail":1,"pl_not_avail":0},{"topic":"P1P2/P/P1P2MQTT/bridge0/C/9/RT","pl_avail":1,"pl_not_avail":0},{"topic":"P1P2/P/P1P2MQTT/bridge0/S/9/Heating_Only","pl_avail":0,"pl_not_avail":1}],"avty_mode":"all","dev":{"name":"HC_Room","ids":["HC_Room"],"mf":"Home-CliCK","mdl":"P1P2MQTT_bridge","sw":"0.9.58rc2"},"ic":"mdi:coolant-temperature","qos":1}
```

Arnold-n, 2024-12-07:
Many thanks, this clearly identifies the cause of the duplicate entries. The Room_Cooling entry is generated when a 00Fx30 packet is received. This packet is generated on my system addressed to auxiliary controller F0 only, but on your system for 3 auxiliary controllers (F0/F1/FF). I will fix this and let you know when the updated version is available.

brainrecursion, 2024-12-07:
Excellent, glad it makes sense!

I also seem to have duplicate entries for quite a few of the UI ones. Not sure if this is same issue?
Thanks
[image]

Arnold-n, 2024-12-07:
Same issue indeed (but a bit trickier as some of the duplicate entries represent different things), and you might also see duplicate field setting values.

Arnold-n, 2024-12-08:
I hope v0.9.58rc4 resolves all duplicate entries and does not introduce any new issues. After installing, `MQTT_Delete_Own_Rebuild` (/D12) is needed to remove/replace some of the old entries.

adam-the, 2025-01-06:
I was also getting duplicates, but only of the HC_Room entries - I didn't think much of it as changing any of them worked anyway haha... 0.9.58rc4 with a rebuild reduced it to one entry.

brainrecursion, 2025-01-08:
I can confirm the rc4 removed my duplicate entries as well.

Arnold-n, 2025-01-08:
Great, thanks for your confirmations.

closed 2025-01-08 by Arnold-n