Arnold-n/P1P2MQTT #154: Reverse Engineering Daikin Ducted AC unit FBQ71C7VEB

yiannosi, 2024-12-23:
Hi,

I ve designed a similar device based on an ESP32 module for the above ducted  split type air conditioner which also has a wired controller with P1P2 interface.

Trying to figure out how the various packets are interpreted, I ve lefet my unit just listening the recording traffic with the AC off and here's what I get :

Note that the first number indicates the length of the packet.

Can anybody help decoding this as it doesn't seem to follow the rules of the P1P2 packets given 

```

new Incoming 13 - 00 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 00 10  - 00 00 62 2 1A 0 11 0 3A 0
new Incoming 11 - 80 0 18 0 0 0 0 0 0   0 8A
new Incoming 13 - 00 00 11- 0 0 0 0 0   13 28 0 0 0
new Incoming 13 - 40 00 11 - 0 -0 -0   -0 -0 -13 -6C -0- 0- 0-
new Incoming 5 - 0 D 20 4 C7
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 5 - 0 E 20 4 34
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 5 - 0 F 20 4 65
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 13 - 0 0 1F 31 40 0 1F 0   76 B 44 C E
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 11 - 80 0 18 0 0 0 0 0 0   0 8A
new Incoming 13 - 0 F0 30 0 23 B0 0   20 10 20 10 1 21
new Incoming 5 - 0 10 20 4 86
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 4 - 80 80 0 12
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 13 - 0 0 20 4 3C 40 0 20   1 23 B0 0 20
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 13 - 0 0 1F 31 40 0 1F 0   76 B 44 C E
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 11 - 80 0 18 0 0 0 0 0 0   0 8A
new Incoming 13 - 0 0 11 0 0 0 0 0 13   28 0 0 0
new Incoming 13 - 40 0 11 0 0 0 0 0   13 6C 0 0 0
new Incoming 5 - 0 1 20 4 6D
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 5 - 0 2 20 4 9E
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 5 - 0 3 20 4 CF
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
new Incoming 13 - 0 0 1F 31 40 0 1F 0   76 B 44 C E
new Incoming 13 - 0 0 10 0 62 2 1A 0   11 0 2 0 0
new Incoming 13 - 40 0 10 0 0 62 2 1A   0 11 0 3A 0
```

Arnold-n, 2024-12-23:
This is standard data for F-series version "C", which is already supported as model 10 by this project. Some requests are replied to (00 00 10, 00 00 11), others are not (80 00 18, 00 00 1F, 00 03 20, etc) and for these extra messages the data is not reverse engineered yet.

yiannosi, 2024-12-23:
> This is standard data for F-series version "C", which is already supported as model 10 by this project. Some requests are replied to (00 00 10, 00 00 11), others are not (80 00 18, 00 00 1F, 00 03 20, etc) and for these extra messages the data is not reverse engineered yet.

Thanks for your quick  reply.
Is there any info available  for the known packets? 
I understand  decoding is done within the p1p2serial library,  correct?

Arnold-n, 2024-12-23:
It's a combination of code in the P1P2MQTT library, P1P2Monitor, and P1P2MQTT-bridge.

yiannosi, 2024-12-23:
> It's a combination of code in the P1P2MQTT library, P1P2Monitor, and P1P2MQTT-bridge.

ok..could you point me where to look for to get some info to decode the packets in my code?

Arnold-n, 2024-12-24:
This code is for ESP8266+ATmega, not ESP32.

closed 2024-12-24 by Arnold-n