Arnold-n/P1P2MQTT #163: HC_Room Room_Heating unavailable in Home Assistant

nicx, 2025-01-08:
I don't know why and since when exactly, but for a few days the HC_Room Room_Heating ist unavailable. Tried to restart the module, rebuild MQTT, but with no luck. All other sensors are working as expected. Any idea what I could do?

[image]

Arnold-n, 2025-01-08:
Unavailability is likely either:
-`P1P2/L/P1P2MQTT/bridge0` being `offline` (should not be the case)
-`P1P2/P/P1P2MQTT/bridge0/A/8/Control_Function` being `0` (`Control_Function` is off)
-`P1P2/P/P1P2MQTT/bridge0/C/9/RT` being `0` (system operating in LWT mode, so no RT control)

If `Control_Function` is off, can you derive when/why this happened?

nicx, 2025-01-08:
@Arnold-n ok, this is interesting: P1P2/P/P1P2MQTT/bridge0/C/9/RT is 0. But looking directly in the Altherma configuration I can see that the mode is set correctly to "room thermostat". I can control the Altherma with my Madoka.

Just tried a complete restart of the Altherma, but the problem is still the same. Any idea?

Arnold-n, 2025-01-08:
Topic `P1P2/P/P1P2MQTT/bridge0/C/9/RT` depends on field setting [C-07], but newer systems do not always communicate field settings except shortly after a restart. So you should see the correct value after a Daikin restart and a 5-minute wait. As long as the field setting is not observed, the "previously seen" field setting is used, or a default setting, which was 0 (which I am now changing to 1 in [P1P2MQTT-bridge v0.9.58rc9 for Daikin E-series](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.58rc9-Daikin-E.ino.bin)).

nicx, 2025-01-08:
> So you should see the correct value after a Daikin restart and a 5-minute wait.

@Arnold-n unfortunately this didn't work here. But with your updated version it works now. Thanks a lot!

closed 2025-01-08 by Arnold-n