Arnold-n/P1P2MQTT #164: BUH sensor always "0"

nicx, 2025-01-12:
I have a 9kW backup heater in my Altherma. 

First question: Why is ther a binary sensor only for "BUH_2"? Am I missing a sensor "BUH_1" or is this name just a default name?
Second question: When my backup heater is running, the binary sensor is showing 0 instead of 1. what could be the problem here? (Imcan see the backup,heater is runninf via ESPAltherma)
Third question: The meters and power sensor for the BUH are not working either. Is this the same problem or another one?

Arnold-n, 2025-01-12:
The BUH_2 sensor is only used on a few systems with 2 BUHs. Can you check whether Gasboiler_Active (originating from the hybrid's system gasboiler) corresponds to your BUH?

If meters are not working, have you enabled Counter_Request_Function?

nicx, 2025-01-13:
@Arnold-n no the Gasboiler_Active does not correspond to the BUH. I dindt find any sensor which correspods. Thats why I am asking ;)

All other meters are working, its only the meter sensor.hc_power_consumption_buh which sticks at 0. the sensor sensor.hc_meters_hours_backup1_dhw on the other hand is showing correct values (about 50h in my case).

Arnold-n, 2025-01-13:
You are right - I think the `BUH_On` status is now indicated by `Climate_Active_Q` (in `HC_Unknown`), I will rename it in a next version.

It is normal that `sensor.hc_power_consumption_buh` remains 0: this depends on an external electricity meter for the BUH, with input to the bridge provided via `P1P2/P/meter/U/9/Electricity_BUH_power`.

nicx, 2025-01-15:
> You are right - I think the `BUH_On` status is now indicated by `Climate_Active_Q` (in `HC_Unknown`), I will rename it in a next version.

hm I think that is nor correct. The sensor "Climate_Active_Q" is always on in my case. So it does not correspon to the BUH. I will do some tests and have a look on all sensors if I could find anything which leads to the right direction...

nicx, 2025-01-15:
@Arnold-n I did some more research: in the HA logbook ther is absolutely no HC sensor which changes the status from 0 to 1 at the time the BUH starts heating. I think its just missing for now. Could I help you with logs to find out  more?

Arnold-n, 2025-01-15:
Thanks - but I think I found it again.. indeed it wasn't any known entity, but an unknown one (thus by default not made available), how about `HC_Mode` / `BUH`, added to [v0.9.58rc11](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.58rc11-Daikin-E.ino.bin)?

nicx, 2025-01-17:
@Arnold-n I have installed the new RC and will give feedback. Thanks!

nicx, 2025-01-26:
> Thanks - but I think I found it again.. indeed it wasn't any known entity, but an unknown one (thus by default not made available), how about `HC_Mode` / `BUH`, added to [v0.9.58rc11](https://github.com/Arnold-n/P1P2MQTT/blob/main/Firmware_images/P1P2MQTT-bridge-v0.9.58rc11-Daikin-E.ino.bin)?

@Arnold-n its working! thanks :)

nicx, 2025-01-26:
@Arnold-n ah I forgot to ask: ist is possible in any way to see which level the backup heater is working? as far as I unsüderstood there are two levels when using the 9kw heater. first level is mostly 3kw, second level is configurable 6kw or 9kw. when knowing the level it would be easy to calculate the power consumption ;)

Arnold-n, 2025-01-26:
Isn't that just BUH_2 (I have only 1 heater level so cannot verify)? If not, that may be another bit not yet found - if you can share (can be by email) hex data log and indicate when the heater switches between modes, I can try to find it.

nicx, 2025-01-26:
@Arnold-n ah ok, I will check that and give feedback.

nicx, 2025-01-28:
@Arnold-n you are right, it's just BUH_2 for the second level. It's now all working as expected! :)

I've created a template helper in Home Assistant for calculating the power consumption of the Altherma including the different levels of the backup heater if anybody is interested in reusing it:

```
{%- set extras = 0 -%}
{%- if is_state('binary_sensor.hc_mode_buh', 'on') %}
  {%- set extras = extras + 3000 %}
{%- endif %}
{%- if is_state('binary_sensor.hc_mode_buh_2', 'on') %}
  {%- set extras = extras + 6000 %}
{%- endif -%}
{{ extras + ( states('sensor.hc_power_consumption_heatpump')  | float ) | round(0, default=0) }}
```

nicx, 2025-03-19:
> It is normal that `sensor.hc_power_consumption_buh` remains 0: this depends on an external electricity meter for the BUH, with input to the bridge provided via `P1P2/P/meter/U/9/Electricity_BUH_power`.

@Arnold-n I recognized that even when publishing values to this topic the sensor sensor.hc_power_consumption_buh always shows "0". As far as I understood you, the sensor should reflecting the value I published to P1P2/P/meter/U/9/Electricity_BUH_power. Am I right?

nicx, 2025-03-24:
@Arnold-n could you help me?

Arnold-n, 2025-03-30:
@nicx can you check that the value of parameter P37 is `P1P2/P/meter/U/9/Electricity_BUH_power` or matches the topic you use, as the topic is configurable, and that the meter reading is published on that topic at least every 75 seconds?

nicx, 2025-03-30:
the value is the same:

[ESP] ESPuptime    352366 P37: mqtt_BUH_Power           is P1P2/P/meter/U/9/Electricity_BUH_Power
[ESP] ESPuptime    352366 P38: mqtt_BUH_Total           is P1P2/P/meter/U/9/Electricity_BUH_Total

[image]

I update the value every 5 seconds.

nicx, 2025-03-30:
Oh shit, is it case sensitive? I just saw I use a topic with a small "power" instead of "Power". Will try that with capital letter.

nicx, 2025-03-31:
Using case sensitive MQTT Topic works. So sorry, my fault.

closed 2025-03-31 by nicx