Arnold-n/P1P2MQTT #171: Feature Request: Make "Flow Target" Editable via User Interface

paqpal, 2025-01-23:
Hi,

It would be great to have the ability to modify the **Flow Target** directly through the user interface. Currently, the system only allows viewing the parameters (8-0b, 8-0c, 8-0d) but not editing them.

**Benefits:**
- **Improved Automations:** Enables creating automations that better maintain the ΔT by adjusting the water flow, thus optimizing system performance.
- **Greater Control:** Helps maintain a more stable ΔT, increasing system efficiency and reliability.
- **User-Friendly:** 

**Additional Consideration:**
It would also be preferable to allow customization of the climate curve in the future. However, this goes beyond the scope of the current request.

Thanks for your attention and the great work!

Paq

Arnold-n, 2025-01-24:
This is on the to-be-added list but I do not think it is a good idea to change this setting dynamically, because any field setting change requires a Daikin system restart. If you want to change the flow dynamically, you can have a look at [intergas_circulation_cap](https://github.com/reinier-vegter/intergas_circulation_cap).

paqpal, 2025-01-24:
Hi

Thank you for your assistance. I was initially considering modifying the pump speed only twice a day, but the solution you suggested appears to be a better approach.

-	I would also like to kindly suggest implementing the unit_of_measurement and device_class for entity HC_15_param Param15_Temperature_0A.

Currently, Home Assistant (HA) displays this parameter incorrectly in the history charts. According to Espaltherma, this parameter represents the refrigerant discharge temperature.

-	Additionally, I’m looking into the issue of the overshoot being stuck at 1°C. A technician from Daikin mentioned that this setting could be increased by configuring the system to heating-only mode. 

However, I need to verify this approach.

-	My next step is to understand the Daikin defrost algorithm and attempt to tweak it, as the factory implementation seems too dumb.

I came across a user who developed an automation that tracks the running time and temperature to predict unnecessary defrost cycles.

This automation restarts the heat pump to reset the running time, with a 3-minute restart period being more effective than initiating a defrost cycle.

I'm not sure if this is within the scope of your project, but I thought it might be of interest.

Thank you again for your support, your device is helping a lot and decreased power consumption by 5kW/h a day.

Best regards,

Paq

Arnold-n, 2025-01-24:
Thanks - I'll add your Param15 info to the param15 work on the todo-list.

Hybrid systems unfortunately have a fixed 1C overshoot which cannot be increased (actually it is an overshoot of max 1.9C due to rounding).

Defrosting has my attention, as the hybrid systems have problems with it, though I am more concerned about the unit not defrosting when it is needed giving COPs down to 1.2. Requesting a defrost manually is already implemented as a HA button. Extending a defrost can only be done by timely off/on switching; not sure whether the cost of that is really less than accepting it. Newer full-electric systems have much better defrosting timing.

paqpal, 2025-01-24:
Well, here in northern Italy lot of people (including me) saw machines defrosting when there's no need.

Temp is around 2°C / -2°C.

Only one time I saw back radiator frosted.
Otherwise I could do my old trick to wrap outside sensor with a tiny foam, shifting outside temp by 2°C

Arnold-n, 2025-11-03:
I've been busy with something else, but it should arrive relatively soon now.

Arnold-n, 2025-11-21:
On my hybrid, I have not (yet) updated the Daikin firmware. On the full-E, Onecta can do the updates via the WLAN adapter in the MMI. The bridge cannot update the Daikin firmware.

What's the issue with the defrost algorithm for you?  On my system defrosts are often too late, so the COP goes down. The bridge has an option to request manually a defrost which may then help. In a future firmware version of the bridge an option will be there to automate defrosts when beneficial.

paqpal, 2025-11-22:
Well we talked about it lots of months ago.
Here in Northern Italy (Pianura Padana) many Daikin units tend to trigger defrost cycles too early and the recovery phase is overly aggressive: circulation pump and heat pump immediately ramp up to maximum speed, which causes new frost to form shortly after.

A Daikin technician suggested limiting the circulation pump to 60 - 70%. After reducing it, the interval between defrost cycles went from about 27 minutes to 35–43 minutes (with outdoor temperatures around –2 °C).
No defrosts after 10 a.m.

On several forums, many users recommend tuning the circulation pump to keep the ΔT to 5°C and the leaving water temperature as low as possible, which helps reduce frost formation.

With this setup, I now have a ΔT of around 5 °C using 10 l/min. My system has large cast-iron radiators, and my highest electricity bill has been €380 for two months (December/January). Electricity is expensive in Italy, but natural gas isn’t exactly cheap either.

What I’ve learned (I hope...) is that defrost cycles disrupt thermal balance quite a lot. By limiting the circulation pump, I (think to) reduce how much that balance is disturbed.

Despite lowering the pump speed, all radiators are still properly supplied and there are no cold rooms.

That’s why I originally asked for the option to adjust parameters like circulation-pump speed and similar settings. It would make these tests much easier to run, without all the hassle.

Thx