Arnold-n/P1P2MQTT #176: Sensor Request: Speed in RPM of outside unit compressor

nicx, 2025-01-28:
Beides the smart grid functionality I have one last sensor which I am missing to get completely rid of the additional ESPAltherma module: It is the speed of the compressor of the outside unit in RPM. Is it possible to have this sensor in P1P2MQTT, too?

Arnold-n, 2025-01-28:
So far I did not find the compressor RPM (or the fan frequency) in the P1/P2 data.

Suedschwede66, 2025-03-23:
HC_15_param Param15_Temperature_02 * 10 = Compressor RPM in U/min

[image]

nicx, 2025-03-24:
@Suedschwede66 cool, thank you!
@Arnold-n could you rename the entity and perhaps directly integrate the factor 10? with that done there would be no need for a helper anymore.

nicx, 2025-03-25:
@Arnold-n a short addition: the sensor value is "-0.1" when the Altherma heating is off. Maybe this could be set to "0" instead.

Arnold-n, 2025-03-30:
@nicx Thanks for your suggestions, implemented in v0.9.58rc16.

nicx, 2025-03-31:
Everything is working fine with the latest RC version. Thanks again @Arnold-n

closed 2025-03-31 by nicx