Arnold-n/P1P2MQTT #177: Sensor Request: Heating flow temperature setpoint value and actual value

nicx, 2025-01-28:
I am missing (or I just didn't find them) the sensors for the heating flow temperature. I don't know exactly how they are called in Englisch, in German it's called the "Vorlauftemperatur", which is the target temperature controlled by the configured heating curve.

I checked the sensor sensor.hc_lwt_lwt_setpoint, but it differs a lot to the ESPAltherma sensor for the LWT set point:

[image]

[image]

For the actual value I found no comparable sensor. Only sensor.hc_unknown_temperature_unknown_q could be a match:

[image]

[image]

Any hints?

forTT78, 2025-01-28:
Hi,
at my installation it's under sensor.hc_sensors_temperature_r2t_leaving_water

Best regards,
Thomas

forTT78, 2025-01-28:
Hi, it's the target temperature and depends on the ambient temperature (e.g. sensor.hc_sensors_temperature_outside_unit).
With out any averaging it will change as function of the outside temperature.

forTT78, 2025-01-28:
@nicx 
> What we do know so far: The "Temperature_R2T_Leaving_Water" sensor is the **current** temperature of the heating water.
> 
> My open main question is: Is the "LWT_Setpoint" sensor is the **target** temperature for the heating water. Is that correct?

Yes.

> In addition to this question my second question is: Why is the sensor value of "LWT_Setpoint" that different to the sensor values of the ESPAltherma pendant (see first two screenshot above)? And why is it updating so slowly in comparison?

From my understanding, the value from ESPAltherma looks more like the (internal) calculated value for target temperature, based on the heating curve and ambient temperature. For "LWT_Setpoint" I can set only whole numbers, no decimal numbers.

But I don't know exactly in which situation daikin uses one of the setpoint values first.

Arnold-n, 2025-01-28:
Indeed LWT_Setpoint is the target. On my hybrid system, it has an accuracy of 0.1C and is equal to ESPAltherma's output. On my EBLA monoblock, it has an accuracy of 1.0C and I do not yet know what ESPAltherma reports. Perhaps the data provided over P1/P2 is rounded by the Daikin unit for some units.

Arnold-n, 2025-01-29:
Mine is an older EHYHBX08AAV3 from 2014, a hybrid heat pump/gas boiler system in cascade. I think your system is an integrated heating/DHW boiler system but not 'hybrid' in the sense of being a heat pump/gas boiler?

nicx, 2025-01-29:
@Arnold-n you are right, it's not "hybrid" in that sense of being. :)

nicx, 2025-03-31:
@Arnold-n I would like to bring this topic up again. what could be the reason that ESPAltherma reports in 0.1 steps, but P1P2 only in 1.0 steps? Is there any way to change this?

BTW: All other P1P2 temperature sensors of my Altherma are reporting in 0.1 steps, only this single one "sensor.hc_lwt_lwt_setpoint" has this other behavior.

closed 2025-05-12 by nicx