Arnold-n/P1P2MQTT #185: Heating Off mode not working - Altherma ELBH12EF9W - Home Assistant

Sonusss, 2025-03-16:
Hi,

Having settled heating to off mode from HP LWT  (climate.hc_lwt_abs_heating) thermostat the heat pump keeps on working.

The Altherma unit is **not** settled to use external contact.

I tried also to set the Altherma to use external thermostat but then got an error on Altherma display about missing thermostat.

Is there a specific "hc" entity to manage general on-off operation ?

Arnold-n, 2025-03-16:
Both the climate entity and the Altherma_On button are to switch the general on-off operation.

Which Daikin system do you have and at which outside temperature did this occur? Sometimes this is normal behaviour due to defrost protection, and in some cases the LWT target is higher in defrost protection mode than it is in normal operation mode.

It could also be running out of write budget of the bridge, which you can resolve by restarting (with button Restart_P1P2Monitor_ATmega), by reducing writes and/or by reducing the Write_Budget_Period (if this is 60, you can do one parameter write/60 minutes, and one on/off action counts as 3 parameter writes).

As for external thermostat, ([C-07] = 2), you can enable the bridge to report a fake sensor temperature with the 'F' command which should resolve the missing thermostat message.

Sonusss, 2025-03-16:
Indoor unit:
ELBH12EF9W
software: 22009C01
EEprom: 1709433-12

Outdoor unit:
ERRA10EV3
Software: 18000211

When I  Restart_P1P2Monitor_ATmega  I got directly  HC_Bridge Error_Budget = 10 normal ?

Outside temp is not very cold, around 10°C

A strange thing is HC_LWT2 LWT_Setpoint_Add is changing a lot without any action on my side...

[image]

Arnold-n, 2025-03-16:
At 10C and not being a monoblock, frost protection is not likely the cause.

An error budget of 10 at ATmega restart, increasing afterwards, is perfectly normal. Only if it decreases, it is a sign of interface issues.

The LWT2_Add shows a weather-dependent target for the 2nd zone and the jumps also likely due to a clock program (day/night difference), looks OK to me. It's not useful unless you have two zones.

Did the restart/write budget solve your issue?

Sonusss, 2025-03-18:
No, the Restart_P1P2Monitor_ATmega didn't fix my issue.
I don't have a second heating zone.

At this stage, I need to check Altherma unit settings but it can't be done before April.
I'm away from home, this is why I'd like to switch heating OFF.

Will come back.

Arnold-n, 2025-03-19:
Even without 2nd heating zone the heat pump still calculates the LWT target, but never uses it.

Do other settings (LWT target change / RT target change) still work? Is Control_Function still ON?

Sonusss, 2025-03-20:
The external unit compressor is OFF for more than two days now but I'm not sure if this is due to control command or due to warm outside temperature conditions in the meanwhile.

It needs to be investigated further when I'm back home...

Sonusss, 2025-04-20:
I made some extra tests and finally it seems that Onecta Cloud is overriding P1P2MQTT ON/OFF command with it's own states

Sonusss, 2025-04-22:
After a while the problem comes back without any Cloud/Internet connectivity.
Looks to be really connected to write budget, but I'm really confused about write parameters...
"HC_Bridge Write_Budget" value is this:
- Number of commands
- Number of bytes transmitted

In the course to allow a dynamic control I settled Write Budget to 12 and Write Budget Period to 30 is it safe ?
I understand that this is made to spare the EEprom but in this case isn't it better to use the external physical switch for ON/OFF control ?

And finally what is the purpose of: HC_Bridge Counter_Request_Function ?

Arnold-n, 2025-04-22:
The write budget is the number of parameter writes. Some actions, like on/off, write to 3 parameters at the same time.

For E-series I do think it is safe to increase the max number of writes - Onecta does not warn at all against too many actions, the only public warning relates to the older airco units. I am using a period of 15 (minutes) myself.

Counter_Request_Function ensures that all counters (kWh/hours) are requested each minute. If not active, reported counter values can quickly become outdated.

Sonusss, 2025-04-23:
Many thanks for your clarification !

closed 2025-04-23 by Sonusss