Arnold-n/P1P2MQTT #188: Sensor Request: Inverter primary current in Ampere

nicx, 2025-03-24:
Is ist possible to get the Inverter primary current in Ampere as a sensor? With that value, it's easy to calculate the current power consumption of the Altherma. I'll do it in the same way with ESPAltherma, where the sensor is existing (register {0x21,0,105,2,-1,"INV primary current (A)"})

This is the current template in HA which could be adopted to P1P2: `{{ ( state_attr('sensor.althermasensors','INV Primärstrom (A)')  | float * 230 | float(0) ) | round(0, default=0) }}`

nicx, 2025-03-24:
I thing I found the corresponding sensor: sensor.hc_15_param_param15_temperature_10 is it! :)

@Arnold-n could you rename it in a future version? thanks!

nicx, 2025-03-24:
@Arnold-n In addition to the renaming of the sensor would it be possible to add an automatically calculated power sensor with the voltage of 230V to P1P2? That would make the HA template sensor and the automation for pushing that value via MQTT to the topic P1P2/P/meter/U/9/Electricity_Power obsolete when not using an external power sensor.

nicx, 2025-03-25:
@Arnold-n a short addition: the sensor value is "-0.1" when the Altherma heating is off. Maybe this could be set to "0" instead.

buennerbernd, 2025-03-26:
@nicx This is a cool observation.

My Altherma 3 H HT is connected by 3 phases.
3 * 230 * Param15_Temperature_10
would rather fit to the power consumption of the house in my case.

nicx, 2025-03-26:
@buennerbernd you are right... should be configurable then, or the template sensor is unavoidable after all.

buennerbernd, 2025-03-26:
I have observed the calculated power over the day. It fits very good to the power consumption of the house.
Perhaps the result of 3 * 230 * Param15_Temperature_10 is 50 or 100 Watt too high, but it is good enough.

With that knowledge I was able to integrate the heat pump into the solar charging tool evcc by MQTT. (evcc is currently leraning to control SG Ready heatpumps.)
Now I'm able to see the consumption there and I would be happy if we would get the SG Ready functionality running, to control it.

See https://docs.evcc.io/en/docs/devices/heating

nicx, 2025-03-26:
@buennerbernd same for me, I currently use EVCC with ESPAltherma for switching the SGready modes. Works great. When the SG feature is working in P1P2 I could get rid of the additional ESPAltherme module :)

DiabloVmax1200, 2025-03-26:
I can confirm this as well for the Geo Variant. The Param15_Temperature_10 seems to be the amps - I only have a single Phase compressor

buennerbernd, 2025-03-27:
Now we have the open question, if the backup heating has its own sensor.

nicx, 2025-03-27:
@buennerbernd no open question, its already answered: https://github.com/Arnold-n/P1P2MQTT/issues/164

Arnold-n, 2025-03-30:
@nicx Thanks for your info - new param15 sensors are added in v0.9.58rc16, and if external meter input is missing, the power consumption will be based on the current info from param15, multiplied by new configurable parameters for voltage (P49, default 230V) and nr of phases (P50, default 1).

nicx, 2025-03-30:
@Arnold-n great, your are the best! will tray it and give feedback.

btw: could you implement a similar calculation for the BUH power consumption? (see https://github.com/Arnold-n/P1P2MQTT/issues/164#issuecomment-2764469486)
In my opinion this should be very easy: we need 2 more configurable parameters (watt usage for buh step 1, and watt usage for buh step 2). in my case its 3000w for step 1, and 6000w for step 2. I have a 9kW buh. With this values configured, they could be just used for the power sensor when the buh sensors are on.

if buh1 is on then buh power consumption is set to 3000w.
if buh2 is on then buh power consumption is set to 3000w+6000w=9000w.

Arnold-n, 2025-03-30:
@nicx Added in v0.9.58rc17. P51/P52 re used to configure power of BUH step1/2. My "3kW" BUH actually uses only ~2.9kW.

nicx, 2025-03-31:
Everything is working fine with the latest RC version. Thanks again @Arnold-n

closed 2025-03-31 by nicx

buennerbernd, 2025-04-01:
@Arnold-n Thank you for that implementation, I have tested this at my Altherma 3 H HT with v0.9.58rc19.

If we had also a C/9/Consumption_Total, it would be even easier to use this directly in evcc.

nicx, 2025-04-03:
@Arnold-n an additional,sensor,for the overall consumption would be great.
@buennerbernd I am using the calc plugin in evcc to sum/add both values of the buh and the heatpump: https://docs.evcc.io/docs/devices/plugins#calc

buennerbernd, 2025-04-04:
@nicx Thank you for that hint, I also had this idea.
The reason why I'm ignoring the Consumption_BUH so for is that the MQTT topics are only updated when they change.
I don't remember if I have configured this by myself.
The Consumption_BUH is most time 0 and is rarely updatet so evcc complains always after a restart that the value is outdated.