Arnold-n/P1P2MQTT #197: Feature Request: Support Altherma 4

joklee, 2025-05-02:
Hi, very nice project.

Do you see a possibility of supporting the Altherma 4? The Altherma 4 also supports Madoka via P1/P2. I'm commissioning an Altherma 4 with ech2O in the next few weeks and could provide support. I'm a software developer, so basic knowledge is a must. :D

[plan.pdf](https://github.com/user-attachments/files/20018172/plan.pdf)

Arnold-n, 2025-05-02:
Would love to see if it works with the Altherma 4 or if the code needs to be modified. From the Altherma 4 manuals it seems that the field settings have been changed completely, and I am not sure how many other changes were made to the protocol. Would be happy to support you if you'd like to try!

joklee, 2025-05-04:
Great, I would love to try it. I'll send you a purchase email right away.

mhuijgen, 2025-08-19:
@joklee Did you managed to get any data from the Altherma 4 yet?

VmaxTurbo, 2025-08-22:
I recently installed the Altherma 4. EPBX10A4V indoor unit, EPSK06AV3 outdoor unit, EKHWSP200D3V3 boiler tank, and Madoka thermostat. The WiFi module via onecta provides me with various sensors in Home Assistant. ESPAltherma doesn't provide me the correct sensors and values.

I have already played around a lot with all (403) known (ESPAltherma) registers, but I have a lot of wrong values for various (needed) sensors like;

Inlet water temp. (R4T): 0
DHW tank temp. (R5T): -40.6
Indoor ambient temp. (R1T): 0.4
Ext. outdoor ambient sensor (R6T): 0

Does anyone have any idea how I can find the correct registers and/or values for an Altherma 4?

The P1P2MQTT would be preferable, because I want being able to switch the DHW heater while energy prices are low to none.