Arnold-n/P1P2MQTT #198: Wrong accuracy of sensor sensor.hc_lwt_lwt_setpoint

nicx, 2025-05-12:
As already recognized [here](https://github.com/Arnold-n/P1P2MQTT/issues/177) there seems a bug in the accuracy of the sensor sensor.hc_lwt_lwt_setpoint. It only shows values in 1.0 steps, but not in 0.1 steps (as all other temperature sensors do).

Why do I think this is a bug? Because when I compare it to the corresponding sensor of LWT2 the behavior differs: sensor.hc_lwt2_lwt_setpoint_add is showing values correctly in 0.1 steps.

sensor.hc_lwt_lwt_setpoint:
[image]

sensor.hc_lwt2_lwt_setpoint_add:
[image]

Arnold-n, 2025-05-16:
It seems some Daikin models only report the setpoints in 1.0C resolution, even though the underlying data is capable of reporting in 0.1C resolution. Not sure if Daikin would call this a feature or a bug...

nicx, 2025-05-17:
@Arnold-n I am just wondering because all other sensors of P1P2 are in 0.1 resolution, and even ESPAltherma reports this explicit value in 0.1 resolution.