Arnold-n/P1P2MQTT #208: ESP crashes WIFI of ASUS router

buennerbernd, 2025-06-08:
Hi Arnold,

unfortunately ASUS routers sometimes struggle with ESP devices. This is also the case with the P1P2 bridge. That means the 2.4 GHz WIFI is broken until the router is restarted.
I have observed that the very last MQTT message that is received is always L_039: offline

Here are some logs out of FHEM before this happened:

```
2025-06-08_08:19:51 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:38
2025-06-08_08:19:52 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-08 08:19:52 Uptime 86340
2025-06-08_08:19:52 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:39
2025-06-08_08:19:54 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:41
2025-06-08_08:19:56 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:43
2025-06-08_08:19:59 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:46
2025-06-08_08:20:00 MQTT2_P1P2_039 M_039_A: 0800201609005C00FFFFFF00E52D0000098E000000000000000000000050C0410000684100A0C3410000000000031100C01400000300000058022602FA00BC02FA00DC004600DC004600AA00F00096005E01012E133F3F0A00000050544A460000000000000000000000000000000000000000000000643C000014000000000001000000000093000000000000000016004004000000001400402C0015E600000000000000000000000000000000000000000000000000000000000000000000000000000000C000000000000000000000000000000000003200140019001400000000000000000000000000000064000064000000000000000000000000000000000000000000000000000000000000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
2025-06-08_08:20:00 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:47
2025-06-08_08:20:02 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-08 08:20:02 Uptime 86350
2025-06-08_08:20:02 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:49
2025-06-08_08:20:04 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:51
2025-06-08_08:20:07 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:54
2025-06-08_08:20:08 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:55
2025-06-08_08:20:10 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:57
2025-06-08_08:20:12 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:19:59
2025-06-08_08:20:12 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-08 08:20:12 Uptime 86360
2025-06-08_08:20:15 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:02
2025-06-08_08:20:16 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:03
2025-06-08_08:20:18 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:05
2025-06-08_08:20:20 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:07
2025-06-08_08:20:22 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-08 08:20:22 Uptime 86370
2025-06-08_08:20:23 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:10
2025-06-08_08:20:24 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:11
2025-06-08_08:20:26 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:13
2025-06-08_08:20:28 MQTT2_P1P2_039 M_1_Param15_09: 140
2025-06-08_08:20:28 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:15
2025-06-08_08:20:29 MQTT2_P1P2_039 A_8_V_Interface: 14.8
2025-06-08_08:20:31 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:18
2025-06-08_08:20:31 MQTT2_P1P2_039 A_8_V_Interface: 14.7
2025-06-08_08:20:32 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-08 08:20:32 Uptime 86380
2025-06-08_08:20:32 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:19
2025-06-08_08:20:34 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:21
2025-06-08_08:20:36 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:23
2025-06-08_08:20:37 MQTT2_P1P2_039 A_8_V_Interface: 14.8
2025-06-08_08:20:39 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:26
2025-06-08_08:20:39 MQTT2_P1P2_039 A_8_V_Interface: 14.7
2025-06-08_08:20:40 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:27
2025-06-08_08:20:42 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-08 08:20:42 Uptime 86390
2025-06-08_08:20:42 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:29
2025-06-08_08:20:44 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:31
2025-06-08_08:20:45 MQTT2_P1P2_039 A_8_V_Interface: 14.8
2025-06-08_08:20:47 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Su 2025-06-08 08:20:34
2025-06-08_08:20:47 MQTT2_P1P2_039 L_039: offline
```

I have noticed that the ESP Uptime is exactly 24h, so I checked the logs 24 hours before:

```
2025-06-07_05:15:54 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 05:15:54 Uptime 248000
2025-06-07_05:15:56 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:43
2025-06-07_05:15:58 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:45
2025-06-07_05:16:00 MQTT2_P1P2_039 M_039_A: 0800201609005C00FFFFFF00E52D0000098E0000000000000000000000C0C541000090410060C9410000000000031100C01400000300000058022602FA00BC02FA00DC004600DC004600AA00F00096005E01012D0F3F3F0A00000050544A460000000000000000000000000000000000000000000000643C000014000000000001000000000093000000000000000015004004000000001400403400161A00000000000000000000000000000000000000000000000000000000000000000000000000000000C000000000000000000000000000000000003200140019001400150000000000000000000000000064000064000000000000000000000000000000000000000000000000000000000000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
2025-06-07_05:16:00 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:47
2025-06-07_05:16:02 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:49
2025-06-07_05:16:04 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:51
2025-06-07_05:16:04 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 05:16:04 Uptime 248010
2025-06-07_05:16:06 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:53
2025-06-07_05:16:08 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:55
2025-06-07_05:16:10 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:57
2025-06-07_05:16:12 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:15:59
2025-06-07_05:16:14 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 05:16:14 Uptime 248020
2025-06-07_05:16:14 MQTT2_P1P2_039 M_9_Date_Time_Daikin: Sa 2025-06-07 05:16:01
[... 66 lines cut ...]
2025-06-07_08:20:56 MQTT2_P1P2_039 name: WiFi_RSSI
2025-06-07_08:20:56 MQTT2_P1P2_039 uniq_id: 039_bridge0_A9_WiFi_RSSI_9
2025-06-07_08:20:56 MQTT2_P1P2_039 stat_t: P1P2/P/039/A/9/WiFi_RSSI
2025-06-07_08:20:56 MQTT2_P1P2_039 A_9_WiFi_RSSI: -58
2025-06-07_08:20:56 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 08:20:56 < ESP reboot reason: wifiManager after WiFi connect failed
2025-06-07_08:20:56 MQTT2_P1P2_039 S_039: * [MON] 2025-06-07 08:20:56 0DE00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFBA
2025-06-07_08:20:56 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 08:20:56 < No ethernet hardware detected
2025-06-07_08:20:56 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 08:20:56 < Connected to WiFi
2025-06-07_08:20:56 MQTT2_P1P2_039 S_039: * [ESP] 2025-06-07 08:20:56 < MQTT client connected
2025-06-07_08:20:56 MQTT2_P1P2_039 stat_t: P1P2/P/039/A/8/Control_Function
2025-06-07_08:20:56 MQTT2_P1P2_039 pl_off: L0
2025-06-07_08:20:56 MQTT2_P1P2_039 uniq_id: 039_bridge0_A8_Control_Function_8
2025-06-07_08:20:56 MQTT2_P1P2_039 pl_on: L1
2025-06-07_08:20:56 MQTT2_P1P2_039 name: Control_Function
2025-06-07_08:20:56 MQTT2_P1P2_039 pl_off: C0
2025-06-07_08:20:56 MQTT2_P1P2_039 stat_t: P1P2/P/039/A/8/Counter_Request_Function
2025-06-07_08:20:56 MQTT2_P1P2_039 name: Counter_Request_Function
2025-06-07_08:20:56 MQTT2_P1P2_039 pl_on: C2
```

Do you have an idea what happened here? Something that could kill the WIFI?
Why is it exactly 24h after the last reboot?
Why is there a 3 hours gap between L_039: offline and online yesterday?

Thank you, Stefan.

Arnold-n, 2025-06-08:
Hi Stefan, do you have any references to the Asus-ESP problems? Not sure what happens here, but I expect it's a problem on the Asus side - even if a client would misbehave, this should not cause the AP to crash.

That "offline" is always the last message is because that message is not transmitted by the ESP as last transmission. If the ESP disconnects too long from the MQTT server, the MQTT server will transmit a message preconfigured by the ESP, a so-called "will message".

buennerbernd, 2025-06-10:
Hello Arnold,

thank you for your answers. I didn't know the concept of last will in MQTT. That explains the offline message.
I agree that this is an ASUS issue. In the internet I found a lot of hints to solve this (changed settings in the router), but nothing really helped.
So far I have set a static IP in the router, but I will try it with a static IP in the bridge. Thank you for that idea!
I will give feedback if this helps a longer time.

buennerbernd, 2025-06-17:
Hi Arnold, regarding the DHCP lease you are right. The default in my router was 1 day. I have extended this to 7 days to reduce the problems with IOT devices.

In the P1P2 bridge I have set a static IP via telnet. I have also aligned the gateway. I hope this was correct:

```
[ESP] 2025-06-17 10:32:11 P03: use static-IP (0=DHCP)   is 1
[ESP] 2025-06-17 10:32:11 P04: static-IP IPv4 address   is 192.168.178.39
[ESP] 2025-06-17 10:32:11 P05: static-IP IPv4 gateway   is 192.168.178.1
[ESP] 2025-06-17 10:32:11 P06: static-IP IPv4 netmask   is 255.255.255.0
```

Before I did this, I have noticed an error in telnet that resulted into a disconnect:

```
[ESP] 2025-06-17 09:35:01 < --(mqtt low mem skip)--
[ESP] 2025-06-17 09:37:02 < --(mqtt low mem skip)--
[MON] 2025-06-17 09:37:28 400000000000014
[MON] 2025-06-17 09:37:29 F130010000000001000000000000000019
[ESP] 2025-06-17 09:37:35 < onMqttDisconnect() reason 0
[MON] 2025-06-17 09:37:35 00001A
[ESP] 2025-06-17 09:37:42 < Reconnect to MQTT server 192.168.178.40:1883 (user /                                                                                                                                                         password *) not successful yet, retrying in 10 seconds
[ESP] 2025-06-17 09:37:44 Uptime 2800, MQTT is disconnected (16 s total 16 s)
[ESP] 2025-06-17 09:38:10 < onMqttDisconnect() reason 0
[MON] 2025-06-17 09:38:10 000000000000000007F01040000000000001A
[ESP] 2025-06-17 09:38:20 Uptime 2836, MQTT is disconnected (52 s total 52 s)
[ESP] 2025-06-17 09:38:20 < Reconnect to MQTT server 192.168.178.40:1883 (user /                                                                                                                                                         password *) not successful yet, retrying in 10 seconds[ESP] 2025-06-17 09:38:36                                                                                                                                                          Trying to connect to fallback MQTT server (first attempt will fail)
[ESP] 2025-06-17 09:38:36 < onMqttDisconnect() reason 0
[MON] 2025-06-17 09:38:36 00000000001
[ESP] 2025-06-17 09:38:57 Uptime 2872, MQTT is disconnected (88 s total 88 s)
[ESP] 2025-06-17 09:38:57 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds
[MON] 2025-06-17 09:38:57 026: 40F131000000010000190611092626AD
[ESP] 2025-06-17 09:38:57 Uptime 2873, MQTT is disconnected (89 s total 89 s)
[ESP] 2025-06-17 09:38:57 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds

[ESP] 2025-06-17 09:39:08 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds
[MON] 2025-06-17 09:39:08 000000045
[ESP] 2025-06-17 09:39:15 Uptime 2891, MQTT is disconnected (107 s total 107 s)
[ESP] 2025-06-17 09:39:15 < onMqttDisconnect() reason 0[MON] 2025-06-17 09:39:15                                                                                                                                                          000000000000007
[ESP] 2025-06-17 09:39:22 Uptime 2897, MQTT is disconnected (113 s total 113 s)
[ESP] 2025-06-17 09:39:22 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds
[MON] 2025-06-17 09:39:22 025: 0000130000D0000000000000000000000000000000000027
[ESP] 2025-06-17 09:39:22 Uptime 2898, MQTT is disconnected (114 s total 114 s)
[ESP] 2025-06-17 09:39:24 Uptime 2900, MQTT is disconnected (116 s total 116 s)
[ESP] 2025-06-17 09:39:32 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds
[ESP] 2025-06-17 09:39:34 Uptime 2910, MQTT is disconnected (126 s total 126 s)
[ESP] 2025-06-17 09:39:40 < onMqttDisconnect() reason 0
[ESP] 2025-06-17 09:39:43 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds
[MON] 2025-06-17 09:39:44 0001240000000000000000000007F01040000000000001A
[ESP] 2025-06-17 09:39:55 Uptime 2930, MQTT is disconnected (146 s total 146 s)
[ESP] 2025-06-17 09:39:55 < Reconnect to MQTT server 192.168.4.12:1883 (user P1P                                                                                                                                                         2/password *) not successful yet, retrying in 10 seconds
[MON] 2025-06-17 09:39:55   0.575: 0000100001010000000016004004000800001400402C0                                                                                                                                                         0BF
[ESP] 2025-06-17 09:39:59 Restarting ESP to attempt to reconnect to Mqtt server
```

After some retry attempts the address of the MQTT server changes to 192.168.4.12:1883
I don't have a MQTT server at this address.

What are the conditions to get the error  `--(mqtt low mem skip)--`?
Can I do something to avoid this?

Thank you for you help!