Arnold-n/P1P2MQTT #211: Setpoint interface always shows 255C

josephwynn-sc, 2025-07-14:
Hello! Thank you for maintaining this project, it has made my life so much better.

Yesterday I updated my bridges' firmware from 0.9.56 to 0.9.58rc21 (P1P2MQTT-bridge-v0.9.58rc21-Daikin-F.ino.bin). Since then, I am having this issue where I cannot set the temperature because HA always interprets the value as 255. I show the problem in this video:

https://github.com/user-attachments/assets/2c2332cb-9428-453d-a6f3-cc8e240d4252

I read the docs and saw something about `HA_Setup`, but I think this is not available for the F-series. I have pressed `MQTT_Rebuild` and this does not fix the issue. Is there something else I'm supposed to do?

The units are Daikin FDYAN140 and FDYBA35.

Arnold-n, 2025-07-14:
Sorry, this must be an issue in the upgrade procedure. You can repair this with the following commands to be given via MQTT topic P1P2/W or with telnet:
-P35 16
-P36 32
-P37 16
-P38 32
-D5
-D0

closed 2025-07-14 by josephwynn-sc

Arnold-n, 2025-07-15:
You may also have to restore P39, the parameter used to select the air-intake temperature instead of the temperature of the controller's sensor, which may have been set incorrectly to 1 if you didn't manually set it to 1:
-P39 0
-D5
-D33

Root cause of this issue found and will be fixed in a newer version.

Thanks for your feedback!

josephwynn-sc, 2025-07-16:
@Arnold-n just confirming: a value of `1` means it will use the air intake sensor And a value of `0` means it will use the controller sensor?

I was changing the value of P39 and I did not see a change in the "field settings" in the controller (my installer told me this was mode 20, param 2).

ArnoldNiessen, 2025-07-16:
>  a value of `1` means it will use the air intake sensor and a value of `0` means it will use the controller sensor?

Yes, but the sensor value is used only as actual measurement value in the climate dial in Home Assistant. It does not change any field setting or operation of the A/C unit.