Arnold-n/P1P2MQTT #220: MQTT delete ALL and rebuild

Koky05, 2025-08-21:
When I hit "MQTT delete all and rebuild" I didn't expect that will delete really all my MQTT messages event from my Zigbee2MQTT communications and from my OctoPrint. Before I hit that I had 32 devices after it, it left only two.
I did it because my realtime COP was stuck for 20 hours.

Arnold-n, 2025-08-21:
Sorry that this happens, but that is the reason the button is disabled unless in setup-mode and it is part of the setup documentation, [5th paragraph](https://github.com/Arnold-n/P1P2MQTT/blob/main/doc/HomeAssistant.md)).

Was your system still operating, and the meter input still available, when the realtime COP was stuck?

Arnold-n, 2025-08-27:
This is intended - your system is heating on the back-up heater (6000W) and not on the compressor. The real-time COP calculation is only done based on (heat produced by compressor) / (electricity used by main unit excluding BUH/BSH).

If temperatures outside are too high, the compressor is severely limited in its operating range, so the system may delay DHW heating, or it may use the backup-heater, depending on how you configured the field settings.

Koky05, 2025-08-27:
Thanks for clarification, now I understand what happening with my heat pump. And sorry for so basic question and mark them as issue.